qiboteam/qibolab #202: Automatic port selection for qq-live

scarrazza, 2022-11-09:
As the title says, we should at least crash if the port is being used.

andrea-pasquale, 2022-11-11:
@scarrazza now for qq-live we already have an automatic port selection mechanism.
If the port it is already used qq-live will check if the next port it is used and it will continue until he finds a port which is available.
See for example here:
https://github.com/qiboteam/qibocal/blob/97550347244aafe9115c75ce4c0c4b29d1cfa029/src/qibocal/cli/_base.py#L78-L82
Is this enough or do you prefer that we simply crash if the port is being used?

scarrazza, 2022-11-11:
Perfect, thanks for pointing this out, I have probably see a testcase with an old qq-live version.

stavros11, 2022-11-13:
Just pointing out that some people had issues with this mechanism. In some cases the loop checking for available ports was hanging and qq-live was never executed. The quick solution I proposed was to use the -port argument to skip the port that was causing the hanging, but I have not checked in detail the issue. There may be a more robust way to select the port automatically.

scarrazza, 2022-11-13:
Right, I believe this problem still there. The code should not hang but at least raise an error saying the port is not available.

scarrazza, 2023-01-23:
@andrea-pasquale shall we close this?

andrea-pasquale, 2023-01-23:
Yeah, I think so @scarrazza. The user can always select a specific port if there are some issues as @stavros11 pointed out.

closed 2023-01-23 by scarrazza