qiboteam/qibolab #208: Platform refactoring for qibocal and new drivers

scarrazza, 2022-11-10:
qibolab should provide abstractions for qibocal platform access.

sorewachigauyo, 2022-11-15:
For the abstract platform setup method (https://github.com/qiboteam/qibolab/blob/main/src/qibolab/platforms/abstract.py#L154-L172), other than setting up  the instruments, this method also creates a mapping of instruments and ports related to readout, qubit control and flux control.

These mappings are currently used in the tiiq calibration to adjust instrument parameters e.g. https://github.com/qiboteam/qibocal/blob/main/src/qibocal/calibrations/characterization/resonator_spectroscopy.py#L58

For new drivers, this assignment requires the readout/control instrument to have a `ports` property which is a dictionary and the flux control instrument to have a `dac` property which is also another dictionary. These properties are not used in the pulse sequence execution of the `MultiqubitPlatform`.

AleCandido, 2023-06-23:
@scarrazza @stavros11 after the last refactoring (and in particular after #469) this issue should be closed, isn't it?

scarrazza, 2023-06-23:
Yes, thanks.

closed 2023-06-23 by scarrazza