qiboteam/qibolab #212: Pulse sequence with two readout pulses fails

andrea-pasquale, 2022-11-14:
I'm working on qiboteam/qibocal#146.
To execute simultaneously a T1 or a T2 experiment on two qubits I'm trying to execute a `PulseSequence` with two `ReadoutPulses`.
However, both in `main` and `alvaro/binning_pr`, the code fails with the following error.
```bash
Traceback (most recent call last):
  File "/nfs/users/andrea.pasquale/test.py", line 29, in <module>
    results = platform.execute_pulse_sequence(sequence, nshots=5000)
  File "/nfs/users/andrea.pasquale/qibolab/src/qibolab/platforms/multiqubit.py", line 41, in execute_pulse_sequence
    acquisition_results.update(self.instruments[name].play_sequence_and_acquire())
  File "/nfs/users/andrea.pasquale/qibolab/src/qibolab/instruments/qblox.py", line 937, in play_sequence_and_acquire
    self.device.get_sequencer_state(sequencer_number, timeout=1)
  File "/home/users/andrea.pasquale/.conda/envs/test/lib/python3.9/site-packages/qblox_instruments/native/generic_func.py", line 531, in decorator_wrapper
    return func(*args, **kwargs)
  File "/home/users/andrea.pasquale/.conda/envs/test/lib/python3.9/site-packages/qblox_instruments/native/cluster.py", line 500, in get_sequencer_state
    return gf.get_sequencer_state(funcs, sequencer, timeout, timeout_poll_res)
  File "/home/users/andrea.pasquale/.conda/envs/test/lib/python3.9/site-packages/qblox_instruments/native/generic_func.py", line 1495, in get_sequencer_state
    raise TimeoutError(
TimeoutError: Sequencer 0 did not stop in timeout period of 1 minutes.
```
Here is the script to reproduce the error:

```python
from qibolab import Platform
from qibolab.pulses import PulseSequence

platform = Platform("tii1q")

platform.connect()
platform.setup()
platform.start()

qd_pulse1 = platform.create_RX_pulse(0, start=0)
qd_pulse2 = platform.create_RX_pulse(1, start=qd_pulse1.finish)
ro_pulse1 = platform.create_qubit_readout_pulse(0, start=qd_pulse2.finish)
ro_pulse2 = platform.create_qubit_readout_pulse(1, start=ro_pulse1.finish+4)
sequence = PulseSequence()
sequence.add(qd_pulse1)
sequence.add(qd_pulse2)
sequence.add(ro_pulse1)
sequence.add(ro_pulse2)

results = platform.execute_pulse_sequence(sequence, nshots=5000)
print(results)

results = platform.execute_pulse_sequence(sequence, nshots=5000)

platform.stop()
platform.disconnect()
```
This could be related to this [issue](https://github.com/qiboteam/qibolab/pull/161#issuecomment-1282192501) that @stavros11 raised in #161, but the error raised is different.
Any ideas on how to solve this @aorgazf @DavidSarlle @maxhant?

aorgazf, 2022-11-14:
Thanks @andrea-pasquale for working on https://github.com/qiboteam/qibocal/issues/146.
I will try to debug the problem asap.

aorgazf, 2022-11-14:
Hi @andrea-pasquale, 
The problem may be on the tii1q runcard, it needs to be updated with the changes of the setup introduced while I was away.
Please try this:
```python
from qibolab import Platform
from qibolab.pulses import PulseSequence

platform = Platform("tii1q")

platform.connect()
platform.setup()
platform.start()

qd_pulse1 = platform.create_RX_pulse(0, start=0)
ro_pulse1 = platform.create_qubit_readout_pulse(0, start=qd_pulse1.finish)

qd_pulse2 = platform.create_RX_pulse(1, start=0)
ro_pulse2 = platform.create_qubit_readout_pulse(1, start=qd_pulse2.finish)
sequence = PulseSequence()
sequence.add(qd_pulse1)
sequence.add(qd_pulse2)
sequence.add(ro_pulse1)
sequence.add(ro_pulse2)

results = platform.execute_pulse_sequence(sequence, nshots=5000)
print(results)

platform.stop()
platform.disconnect()
```

andrea-pasquale, 2022-11-14:
Thanks @aorgazf.

andrea-pasquale, 2022-11-14:
Still not working, I'll try again later with the updated runcard @aorgazf.

aorgazf, 2022-11-30:
This issue was fixed in the latest version of [alvaro/binning_pr](https://github.com/qiboteam/qibolab/tree/alvaro/binning_pr)

closed 2022-11-30 by aorgazf