qiboteam/qibolab #214: Pulse Amplitude Validation

sorewachigauyo, 2022-11-15:
In https://github.com/qiboteam/qibolab/blob/main/src/qibolab/pulses.py#L411-L412, there is validation for the amplitude of the pulse reading to [-1, 1]. Currently for the IcarusQ RFSoC we are programming the pulses with the 14-bit DAC values, so this validation fails for our instrument. 

Maybe this validation can be moved to the instrument/driver level instead.

stavros11, 2022-11-15:
Same for #211. @maxhant proposed to normalize amplitudes using the maximum possible amplitude value of the device so that everything is within [-1, 1]. This requires some additional calculations in the driver and storing the max_value in the runcard so I am not sure if it is preferrable.

aorgazf, 2022-11-16:
Hi @sorewachigauyo,
@Jacfomg told me that ZI pulse amplitudes are also from -1 to 1. @stavros11, do you know how does QM pulses work?
I am slightly more inclined to normalising the amplitudes so that everything above platform is platform agnostic, but perhaps is my qblox bias :)

stavros11, 2022-11-16:
> @stavros11, do you know how does Quantum Machines pulses work?

I have not started working on this yet but from a quick look at their docs and public repositories (for example the amplitude Rabi experiment [here](https://docs.qualang.io/libs/examples/calibration/rabi-sweeps/)) it seems that they are using amplitudes smaller than 1.

sorewachigauyo, 2022-11-17:
> I am slightly more inclined to normalising the amplitudes so that everything above platform is platform agnostic, but perhaps is my qblox bias :)

I can make these changes on the driver, but if we are going this route, we should fix the description of the amplitude. Currently the description of the amplitude is
```
amplitude (float): Pulse digital amplitude (unitless) [0 to 1].
```

I think that the attributes of the Pulse object are already platform and driver agnostic, but the values need not be.

sorewachigauyo, 2022-12-07:
On the subject of pulses, there is also validation for the start time and duration of the pulses to be integers. Currently I think this is the case due to the 1GSps sampling rate of the Qblox.

Maybe this can include float as well as there are instruments with different sampling rates.

closed 2022-12-14 by scarrazza