qiboteam/qibolab #226: Add tests that measure the performance of drivers

aorgazf, 2022-11-23:
Changes to the driver code can affect its performance inadvertently.
In order to be able to determine the positive or negative impact of a change, we could implement a set of speed tests that measured parameters like:
- [ ] total execution time for predefined sets of circuits and number of hardware averages.
- [ ] overhead added by the drivers

The test could either fail if performance is degraded beyond a predefined threshold or warning if negative but below the threshold and pass on performance improvement.

Similar tests should then be conducted from qibocal to assess how much overhead is added by it.

scarrazza, 2022-11-24:
This shares profiling/timing tools useful for #153.

AleCandido, 2023-12-18:
https://asv.readthedocs.io/en/v0.6.1/