qiboteam/qibolab #248: Pulse duration validation

sorewachigauyo, 2022-12-14:
Currently the pulse duration and start parameters are in integers of nanoseconds. However, this is an issue for instruments with sampling rates that are not 1GSps.

For the IcarusQ RFSoC, we are using a [bisection](https://github.com/qiboteam/qibolab/blob/icarusq_multiqubit/src/qibolab/instruments/icarusqfpga.py#L263-L268) to retrieve the sample indices associated with the pulse.

Perhaps this parameter can be sanitized on the driver side instead?

stavros11, 2023-11-07:
Is this issue still relevant? Note that the constraint for the pulse duration to be integer of nanoseconds was lifted in #623.

sorewachigauyo, 2023-11-14:
I think this issue is still a problem, on commit `f08149e39855d502f8d270d2c2c8efb30c801579`

Trying this code
```
from qibolab.pulses import Pulse

Pulse(start=5, duration=34.33, amplitude=1, frequency=5e9, relative_phase=0, shape="Rectangular", channel="x1", qubit=0)
```
raises the following error
```
  File "c:\Users\745 G5\Documents\repos\icarusq_runcard\test.py", line 4, in <module>
    Pulse(start=5, duration=34.33, amplitude=1, frequency=5e9, relative_phase=0, shape="Rectangular", channel="x1", qubit=0)
  File "C:\qibolab\src\qibolab\pulses.py", line 704, in __init__
    self.duration = duration
    ^^^^^^^^^^^^^
  File "C:\qibolab\src\qibolab\pulses.py", line 808, in duration
    self._finish = self._start + self._duration
                   ~~~~~~~~~~~~^~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'
```

stavros11, 2023-11-14:
Thanks for checking @sorewachigauyo. Should be fixed in #660.

PR #660 (2023-11-15): Allow float duration and start time in pulses
Fixes #248.

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [ ] Tests are passing.
- [ ] Coverage does not decrease.
- [ ] Documentation is updated.
closed 2023-11-15 by stavros11