qiboteam/qibolab #302: Document how to define new platforms using the `InstrumentDesign` class

stavros11, 2023-02-07:
Maybe useful to use mermaid format https://mermaid.js.org/ to make a flowchart. There is also a [Sphinx integration](https://github.com/mgaitan/sphinxcontrib-mermaid) to render the diagrams.

_Originally posted by @AleCandido in https://github.com/qiboteam/qibolab/pull/259#discussion_r1066754359_

scarrazza, 2023-02-07:
Indeed, looks like an useful tool for this task.

AleCandido, 2023-06-23:
This problem does not exist any longer, since the `InstrumentDesign` has been abandoned (well, we could rephrase in terms of the new `Platform` and the instruments themselves, but this makes it equivalent to document the `create` format, that is just part of an overall need for documentation improvement).

closed 2023-06-23 by AleCandido