qiboteam/qibolab #306: Enhancements for QM driver

stavros11, 2023-02-07:
1. https://github.com/qiboteam/qibolab/pull/259#discussion_r1066875355: dataclass for config
2. https://github.com/qiboteam/qibolab/pull/259#discussion_r1066878995: connection status: could be relevant for other drivers
3. https://github.com/qiboteam/qibolab/pull/259#discussion_r1066883787: connection status: could be relevant for other drivers
4. https://github.com/qiboteam/qibolab/pull/259#discussion_r1066891980: dataclass for registering pulses
5. https://github.com/qiboteam/qibolab/pull/259#issuecomment-1411546502: bug when playing more than on flux pulses on the same channel, partially fixed in #295 but we need to use baking in a smarter way to avoid waiting >16ns before different flux pulses

scarrazza, 2023-08-10:
Shall we close this issue?

stavros11, 2023-10-31:
Closing this as the important features are already implemented and the rest is not very relevant or priority.