qiboteam/qibolab #323: SymbolicExpression and large number of pulses

sorewachigauyo, 2023-03-07:
For a typical RB experiment, we iterate over a large number of pulses/gates. While doing this, I noticed that the pulse creation and pulse sequence creation was slowing down badly when we were reaching 50+ pulses. Mainly, it was freezing during the initialization of the symbolic expressions. This may be due to the garbage collection not working as intended and long lookup in the regex.

I noticed that there was also a static method `SymbolicExpression.clear_instances()` and using that after each pulse sequence creation sped up the code.

Example code

```python
import numpy as np
from qibolab.pulses import Pulse, PulseSequence

num_gates = np.arange(1, 51, 3)

for idx, ngates in enumerate(num_gates):
    array = []
    start = 5

    for _ in range(ngates):
        array.append(Pulse(
            start=start,
            duration=1,
            amplitude=0.2,
            frequency=5e9,
            relative_phase=0,
            shape="Rectangular",
            channel="x1",
            qubit=0
        ))
        start += 1
    
    ps = PulseSequence(*array)
    #SymbolicExpression.clear_instances()
```

Timing comparison graph
[image]

sorewachigauyo, 2023-03-09:
Also, I forgot to mention this in the meeting, but it appears that `SymbolicExpression` is not thread-safe.

I tried to delegate the pulse creation to thread workers to speed up the experiment but it fails since the SE lookup table is being modified between the threads.

AleCandido, 2023-03-09:
> Also, I forgot to mention this in the meeting, but it appears that `SymbolicExpression` is not thread-safe.

I know it is not, it is rather simple: if you have a look at the definition, you'll find that it creates a global cache of symbols. And of course, global variables are not thread safe.

https://github.com/qiboteam/qibolab/blob/7a9fe31635ed31f54eeb1dd79809814a5f0fc037/src/qibolab/symbolic.py#L5-L16
https://github.com/qiboteam/qibolab/blob/7a9fe31635ed31f54eeb1dd79809814a5f0fc037/src/qibolab/symbolic.py#L508

aorgazf, 2023-03-15:
Hi @sorewachigauyo,
Thanks for raising this issue.
When I added support for symbolic expressions in Pulses, I did not think it could cause such performance impact.
I've tried to reproduce your results:
``` python
import numpy as np
import matplotlib.pyplot as plt
import timeit
from qibolab.symbolic import floatSymbolicExpression as se_float
from qibolab.symbolic import intSymbolicExpression as se_int
from qibolab.symbolic import SymbolicExpression
from qibolab.pulses import Pulse, PulseSequence

def create_sequence(ngates):
    array = []
    start = 5

    for _ in range(ngates):
        array.append(Pulse(
            start=start,
            duration=1,
            amplitude=0.2,
            frequency=5e9,
            relative_phase=0,
            shape="Rectangular",
            channel="x1",
            qubit=0
        ))
        start += 1
    
    ps = PulseSequence(*array)
    #SymbolicExpression.clear_instances()

t = []
SymbolicExpression.clear_instances()
ngates_list = np.arange(1, 53, 3)
for ngates in ngates_list:
    t.append(timeit.timeit("create_sequence(ngates)", globals=globals(), number=1))

plt.plot(ngates_list, t)
```
[image]

You may notice I get better results. I suspect there were already many symbolic expressions saved in the instances dictionary when you ran your test, and that affected the results.
A way to improve on this would be to call `SymbolicExpression.clear_instances()` once, just before beginning the test.

Nonetheless, I ran the test for higher number of gates, and the time does not grow linearly with the number of pulses.
[image]

I suppose we have the following options:
   - recode the `evaluate()` function of the symbolic expressions and improve its performance. There is definitely room for improvement.
   - do not evaluate the expression unless it is required by the user. 
   - remove support for symbolic expressions in Pulses
   - allow the users to decided whether they want Pulses with symbolic expressions or not 

I will wait to hear your feedback and that from the rest of the team before implementing any of the above.

Regarding the issue of se not being thread-safe, there needs to be a global dictionary where all the symbolic expressions are stored. This dictionary is required for resolving the expressions later on.
I think this can be solved with `from threading import Lock`

AleCandido, 2023-03-15:
> Regarding the issue of se not being thread-safe, there needs to be a global dictionary where all the symbolic expressions are stored. This dictionary is required for resolving the expressions later on.
> I think this can be solved with `from threading import Lock`

If you have a `Lock` you can actually make it thread safe, but at a huge price every access will have an overhead to acquire the lock, and you lose the advantage of threads since you're synchronizing them (so you're partially restoring a sequential execution).

If you want to support threads, we should redesign the class (but Python is not optimal for multithreading in the first place, so better to make use of external libraries).

However, I'd like to understand the goal of this structure, because:
- it might be simplified, e.g. it reimplements part of what Sympy is already doing, mainly to implement lazy evaluation, that is already implemented by performant libraries like TensorFlow
- we might use a different strategy to achieve the same goal

But before spending much time and effort, I would try to make a plan together, to make it clear what is the task and try to achieve in a minimal way (as much as possible, of course).

closed 2023-05-03 by aorgazf