qiboteam/qibolab #328: Pulse deep_copy

rodolfocarobene, 2023-03-14:
PulseSequence has a deep_copy function that calls a pulse deep_copy function, however I don't think that the pulse.deep_copy function is actually implemented somewhere

https://github.com/qiboteam/qibolab/blob/7a9fe31635ed31f54eeb1dd79809814a5f0fc037/src/qibolab/pulses.py#L1473-L1479

AleCandido, 2023-03-15:
Concerning the comment about the static analyzer: unfortunately PyLint is complaining about a lot of things, but it is quite ignored.
https://github.com/qiboteam/qibolab/actions/runs/4363029412/jobs/7628619361#step:5:13

However, it is not complaining about this one in particular, because it doesn't know the type of `self.pulses`, so it can't determine at "analysis-time" (compile-time equivalent) that that method will not exist:
https://github.com/qiboteam/qibolab/blob/7a9fe31635ed31f54eeb1dd79809814a5f0fc037/src/qibolab/pulses.py#L1332-L1334

You can solve this by declaring the type:
```py
from typing import List

# ...

class PulseSequence:
    def __init__(self, *pulses):
        self.pulses: List[Pulse] = []
        self.add(*pulses)
```

AleCandido, 2023-03-15:
An even better option would be to just mark the type of the input type:
```py
class PulseSequence:
    def __init__(self, *pulses: Pulse):
        self.pulses = []
        self.add(*pulses)
```

In this way, even the user of `PulseSequence` would be warned by the analyzer if something else is passed.

In general, I wanted to propose an incremental adoption of type hints in Qibolab and Qibocal, but I was postponing for a while, to first digest dataclasses (that already make use of types, but with runtime semantic, that is more familiar to the average Python dev).
The good thing about type hints in Python (similar to TypeScript) is that you don't need to put everywhere to start, so you can really add them incrementally.

AleCandido, 2023-03-15:
At a deeper inspection, last proposal is impossible, because of the semantic of `PulseSequence.add()`, and consequently the constructor as well.

The correct one should be:
```py
from typing import List, Union

# ...
class PulseSequence:
    def __init__(self, *pulses: Union[Pulse, Self]):
        self.pulses: List[Pulse] = []
        self.add(*pulses)
```
Unfortunately `Self` is only available since Python 3.11, so the only option actually viable is the first one (just add the type to the attribute).

aorgazf, 2023-03-20:
Hi @AleCandido, thanks for your advice. I did in the past make use of type hints, but some of the most advanced features (at the time) required python 3.10.
I personally do not like `from typing import List, Union`. As you point out, in the newer versions of python, this library is not required. As soon as we move to these versions I will add more type hints.

commit a0f6938d (2023-03-22): Merge pull request #336 from qiboteam/alvaro/fix_pulsesequence_deepcopy

Fix call to Pulse.copy() method, closes #328
closed 2023-03-22 by scarrazza