qiboteam/qibolab #329: Multiplex not working for QBlox

maxhant, 2023-03-14:
Hi, 

I am not sure why, but it seems that QBlox no longer multiplexing with hardware demodulation. Here is the error:
```
[33;20m[Qibocal 0.0.2.dev0|WARNING|2023-03-14 10:21:08]: Deleting previous directory debug.[0m
[Qibo 0.1.10|INFO|2023-03-14 10:21:08]: Loading platform qili1q_os2 from runcard debug/new_platform.yml
[Qibo 0.1.10|INFO|2023-03-14 10:21:11]: Using qibolab (qili1q_os2) backend on /CPU:0
[Qibo 0.1.10|INFO|2023-03-14 10:21:11]: Connecting to qili1q_os2 instrument cluster.
[Qibo 0.1.10|INFO|2023-03-14 10:21:13]: Connecting to qili1q_os2 instrument qrm_rf0.
[Qibo 0.1.10|INFO|2023-03-14 10:21:14]: Connecting to qili1q_os2 instrument qrm_rf1.
[Qibo 0.1.10|INFO|2023-03-14 10:21:14]: Connecting to qili1q_os2 instrument qcm_rf2.
[Qibo 0.1.10|INFO|2023-03-14 10:21:14]: Connecting to qili1q_os2 instrument qcm_rf3.
[Qibo 0.1.10|INFO|2023-03-14 10:21:14]: Connecting to qili1q_os2 instrument qcm_bb1.
[Qibo 0.1.10|INFO|2023-03-14 10:21:15]: Connecting to qili1q_os2 instrument qcm_bb2.
Traceback (most recent call last):
  File "/nfs/users/maxime.hantute/qibocal/src/qibocal/cli/builders.py", line 253, in execute
    parser = niGSCactionParser(self.runcard, self.folder, action)
  File "/nfs/users/maxime.hantute/qibocal/src/qibocal/cli/builders.py", line 69, in __init__
    self.nqubits = self.runcard["actions"][self.name]["nqubits"]
KeyError: 'nqubits'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/users/maxime.hantute/.conda/envs/test/bin/qq", line 8, in <module>
    sys.exit(command())
  File "/home/users/maxime.hantute/.conda/envs/test/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/users/maxime.hantute/.conda/envs/test/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/users/maxime.hantute/.conda/envs/test/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/users/maxime.hantute/.conda/envs/test/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/maxime.hantute/qibocal/src/qibocal/cli/_base.py", line 59, in command
    builder.execute()
  File "/nfs/users/maxime.hantute/qibocal/src/qibocal/cli/builders.py", line 260, in execute
    parser.execute(self.format, self.platform, self.qubits)
  File "/nfs/users/maxime.hantute/qibocal/src/qibocal/cli/builders.py", line 55, in execute
    for data in results:
  File "/nfs/users/maxime.hantute/qibocal/src/qibocal/calibrations/characterization/resonator_spectroscopy.py", line 116, in resonator_spectroscopy
    results = platform.execute_pulse_sequence(sequence)
  File "/nfs/users/maxime.hantute/qibolab/src/qibolab/platforms/multiqubit.py", line 228, in execute_pulse_sequence
    results = self.instruments[name].acquire()
  File "/nfs/users/maxime.hantute/qibolab/src/qibolab/instruments/qblox.py", line 1159, in acquire
    binned_raw_results[acquisition_name], pulse, demodulate=False
KeyError: 'ReadoutPulse(0, 2000, 0.4, 140_000_000, 0, Rectangular(), L3-27_a, B1)'
```
I've used the debug mode to understand what was the problem, and it seems to be due to how the data is retrieved. It is wrongly associating all RO pulses to `sequence.number = 0`, whereas it should have only one RO pulse per sequencer. 

Only doing `binned_raw_results = self.device.get_acquisitions(1)` would give me the results for `KeyError: 'ReadoutPulse(0, 2000, 0.4, 140_000_000, 0, Rectangular(), L3-27_a, B1)'`

https://github.com/qiboteam/qibolab/blob/7a9fe31635ed31f54eeb1dd79809814a5f0fc037/src/qibolab/instruments/qblox.py#L1154-L1160

To replicate with my runcard, you can use `qibocal` main on the qili1q_os2 (queues not up to date, and I could not change their name) with the `qibolab` `maxime/runcard_25Q_B`

andrea-pasquale, 2023-03-16:
I was also able to reproduce the issue.
As @maxhant said this is the main problem:

> It is wrongly associating all RO pulses to `sequence.number = 0`, whereas it should have only one RO pulse per sequencer.

The code uses often the attribute `DEFAULT_SEQUENCERS`

https://github.com/qiboteam/qibolab/blob/7a9fe31635ed31f54eeb1dd79809814a5f0fc037/src/qibolab/instruments/qblox.py#L374

which is by default 0 for all ports.
A naive solution could be to rid of this attribute, but I don't know whether this could generate side effects.

aorgazf, 2023-03-16:
Thanks @maxhant for raising this issue.
The bug was in multiqubit, the driver is fine.
I will raise a PR with the patch.

closed 2023-03-21 by scarrazza