qiboteam/qibolab #366: KeyError for circuits with virtual gates in qblox

vodovozovaliza, 2023-04-06:
**Describe the bug**
When the circuit with virtual gates (such as `gates.I` or `gates.RZ`) is transpiled into pulses and sent to the `qblox.Sequencer`, a `KeyError` occurs as the frequency of the `pulse.serial` of the Readout pulse and the key in the `binned_raw_results` dictionary do not correspond.

https://github.com/qiboteam/qibolab/blob/7a2d4c0d6cd8bd4908f63bc38d139ee0cfab52ea/src/qibolab/instruments/qblox.py#L1152-L1157

**To Reproduce**

1. Checkout `jadwiga/transpiler_changes2` that inherits `alvaro/qw5q_runcard_april23` branch with the #362 issue fixed

2. Generate a Qibo circuit with `gates.I` and get the samples
```python
import qibo
from qibo import gates
from qibo.models import Circuit

c = Circuit(5)
c.add(gates.I(0))
c.add(gates.U3(0, 0, 0, 0))
c.add(gates.M(0))

# Execute the circuit on hardware
qibo.set_backend("qibolab", platform="qw5q_gold_qblox")

for _ in range(5):
    result = c(nshots=5)
    print(result.probabilities())
```

4. Create an `run_job.sh` file that would run the code. 
```bash
#!/bin/bash
#SBATCH --partition=qw5q_gold

python qblox_circuit.py
```

5. Run `sbatch run_job.sh`

6. The following error occurs
```python
Traceback (most recent call last):
  File "/home/users/yelyzaveta.vodovozova/qibocal-1/qblox_circuit.py", line 14, in <module>
    result = c(nshots=5)
  File "/nfs/users/yelyzaveta.vodovozova/qibo/src/qibo/models/circuit.py", line 1010, in __call__
    return self.execute(initial_state=initial_state, nshots=nshots)
  File "/nfs/users/yelyzaveta.vodovozova/qibo/src/qibo/models/circuit.py", line 1006, in execute
    return GlobalBackend().execute_circuit(self, initial_state, nshots)
  File "/home/users/yelyzaveta.vodovozova/.conda/envs/rb-env/lib/python3.9/site-packages/qibolab/backends.py", line 93, in execute_circuit
    readout = self.platform.execute_pulse_sequence(sequence, nshots)
  File "/home/users/yelyzaveta.vodovozova/.conda/envs/rb-env/lib/python3.9/site-packages/qibolab/platforms/multiqubit.py", line 227, in execute_pulse_sequence
    results = self.instruments[name].acquire()
  File "/home/users/yelyzaveta.vodovozova/.conda/envs/rb-env/lib/python3.9/site-packages/qibolab/instruments/qblox.py", line 1157, in acquire
    binned_raw_results[acquisition_name], pulse, demodulate=False
KeyError: 'ReadoutPulse(80, 2000, 0.4, 7_212_407_558, 0, Rectangular(), L3-25a, 0)'
``` 

These are the keys of the `binned_raw_results` dictionary from above
```python
['ReadoutPulse(80, 2000, 0.4, -37_592_442, 0, Rectangular(), L3-25a, 0)', 'scope_acquisition']
```

**Expected behavior**
The frequencies of the ReadoutPulse correspond and the pulse sequence is executed.

andrea-pasquale, 2023-04-06:
@vodovozovaliza I tested your code with main using the Alvaro's runcard from `alvaro/qw5q_runcard_april23` and it does not raise the error. I fixed the error in #335.
The problem is that `alvaro/qw5q_runcard_april23` doesn't contain the fix since it is currently 74 commits behind main.

From my side we can merge `alvaro/qw5q_runcard_april23` to main so that everyone is able to access the runcard with the fix already implemented.

andrea-pasquale, 2023-04-06:
@vodovozovaliza #355 has been merged to main.
Merging `main` in `jadwiga/transpiler_changes2` should fix the issue.

vodovozovaliza, 2023-04-06:
Thank you @andrea-pasquale. Merging `main` fixed the bug.

closed 2023-04-06 by vodovozovaliza