qiboteam/qibolab #380: Several transpilation INFO while executing several circuits

MatteoRobbiati, 2023-04-18:
I think in some situations it can be nice to remove the circuit transpilation INFOs. For example, while running quantum machine learning stuff, several circuit executions are required, leading to something like this in the slurm output:

```
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:57]: Circuit can be executed.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:58]: Transpiling circuit.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:58]: Circuit can be executed.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:58]: Transpiling circuit.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:58]: Circuit can be executed.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:58]: Transpiling circuit.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:58]: Circuit can be executed.

(...)

[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:59]: Transpiling circuit.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:59]: Circuit can be executed.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:59]: Transpiling circuit.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:59]: Circuit can be executed.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:59]: Transpiling circuit.
[Qibo 0.1.12.dev0|INFO|2023-04-18 15:39:59]: Circuit can be executed.
``` 

This makes reading the file rather annoying. 
Surely this problem can be eliminated by setting e.g. `QIBO_LOG_LEVEL=4`, but more in general I was wondering about the usefulness to have these info messages.

AleCandido, 2023-06-23:
@MatteoRobbiati @Simone-Bordoni this problem should be pretty easy to solve or close:

```
❯ rg 'log\.'
gate_decompositions.py
385:            log.info(message)

abstract.py
97:        log.info("Transpiler test passed.")

routing.py
245:                log.info(

star_connectivity.py
53:            log.info(message)

pipeline.py
27:            log.info("Transpiling circuit.")
```

Since these are the occurrences of the logger. If you believe it's relevant at INFO level, then we should maybe just consider to change default log level in Qibo. Otherwise, just remove all the few not relevant `log.info()` calls.

Simone-Bordoni, 2023-06-23:
I will remove some log, maybe using errors like what i did for routing and placement. star connectivity will probably be eliminated soon. anyway for QML you should run the transpiler once, it is just the parameters changing...

closed 2023-10-10 by scarrazza