qiboteam/qibolab #385: Raise test failure for self-runner pytests

scarrazza, 2023-04-25:
As discussed #384, we do not have a mechanism to determine if a pytest output executed on a self-runner has failed and e.g. show this info the actions page.

AleCandido, 2023-04-26:
https://docs.pytest.org/en/7.1.x/how-to/output.html#creating-resultlog-format-files

AleCandido, 2023-04-26:
There are actually dedicated actions to parse 2. and integrate with GitHub, I would suggest actually going in that direction.

Not sure about which is the *optimal* one (most likely we should just try), but a few sensible candidates are:
1. https://github.com/marketplace/actions/test-reporter
2. https://github.com/marketplace/actions/publish-test-results
3. https://github.com/marketplace/actions/junit-report-action

Or even applications:
- https://github.com/marketplace/check-run-reporter
- https://github.com/marketplace/testquality
- ...

but I would avoid them, since they mostly offered paid plans, and an action seems simpler to integrate in a workflow.

AleCandido, 2023-05-01:
Since it's possible, and even potentially simple, I would keep this open to also implement the report upload.

If you prefer, feel free to close again and replace with a separate issue, mentioning this one.

closed 2023-10-10 by scarrazza