qiboteam/qibolab #388: Different Drag pulses definiton between Qibolab and Qick

rodolfocarobene, 2023-04-26:
@andrea-pasquale was testing the board and discovered that sometimes drag pulses fail (for large beta parameters).
We now know that drag pulses for the RFSoC don't completely work since they are defined in different ways in qibolab and in qick.

In particular:
Qibolab
https://github.com/qiboteam/qibolab/blob/59360c1eacaef084d895b3043248ee2eaf5c3b2f/src/qibolab/pulses.py#L285-L321

Qick (https://github.com/openquantumhardware/qick/blob/b6604c1228feedd82374802711183da2a332b6ea/qick_lib/qick/helpers.py#L31-L57)

```python
def DRAG(mu, si, length, maxv, delta, alpha):
    """
    Create I and Q arrays for a DRAG pulse.
    Based on QubiC and Qiskit-Pulse implementations.
    :param mu: Mu (peak offset) of Gaussian
    :type mu: float
    :param si: Sigma (standard deviation) of Gaussian
    :type si: float
    :param length: Length of array
    :type length: int
    :param maxv: Maximum amplitude of Gaussian
    :type maxv: float
    :param delta: anharmonicity of the qubit (units of 1/sample time)
    :type delta: float
    :param alpha: alpha parameter of DRAG (order-1 scale factor)
    :type alpha: float
    :return: Numpy array with I and Q components of the DRAG pulse
    :rtype: array, array
    """
    x = np.arange(0, length)
    gaus = maxv * np.exp(-(x-mu)**2/si**2)
    # derivative of the gaussian
    dgaus = -(x-mu)/(si**2)*gaus
    idata = gaus
    qdata = -1 * alpha * dgaus / delta
    return idata, 
```

I was just setting in the driver `delta = 1` but this is not correct.

@andrea-pasquale found that `alpha = - beta` and that `delta=0.5 * sampling_rate / 1e9` but this is a bit weird since `delta` should be the anharmonicity frequency (from qick comments). 
Also, there seems to be some factors 2 in the `sigma` definition...

@igres26, @stavros11, if you have some suggestion on why the definition of the pulse are different and how are they related, I could fix it easily. Otherwise I think I can, in qibosoq, add the pulses from the IQ values, but this may be a bit more difficult.

rodolfocarobene, 2023-05-01:
Now the problem should be solved in qibosoq, in the sense that now I'm setting delta equal to 1 in the correct way.

However, it may be interesting to add delta as a parameter also in qibolab. 
I think that drag pulses are, in theory, anharmonicity dependant and having the same dependance in qibolab could help to reduce the leakage (and we would also be using a physical parameter).

andrea-pasquale, 2023-05-01:
@rodolfocarobene we still need to fix the normalization of the drag pulse, correct?
You remember that when we running allXY with `beta=0` we got some strange results compare to the allXY with the Gaussian pulse.

rodolfocarobene, 2023-05-01:
Yes you are right... I forgot about that. I'll look into that

andrea-pasquale, 2023-05-01:
We are not in a hurry :)
I just suggest to keep this issue open until we solve all the problems related to the Drago pulse.

rodolfocarobene, 2023-06-09:
So, I report here my recent discoveries.
I now have `alpha=beta` and `delta=-LN` so that the two definitions are equal (LN is a large number to which delta is internally divided).

However, looking at raw acquisition of drag pulses, the differences are minimal/non-existing for `beta` around 1, while it's possible to see some differences when  beta is around 100.

If I set `delta=1`, things will probably improve a lot, but at this point maybe the definition of drag pulse that we have in the documentation is not complete...

In any case, I am waiting for testing things on hardware

rodolfocarobene, 2023-06-26:
I'm closing it since now qibosoq uses the same definition of qibolab :-).

closed 2023-06-26 by rodolfocarobene