qiboteam/qibolab #402: Sweeper ordering

Jacfomg, 2023-05-02:
I have issues with how do I order the sweepers on the ZI driver. Frequency sweeps to be done in 2D scans need to be on the outer loop so they are executed sequentally unlike the inner loop which will be swept ciclycally. This is due to hardware limitations and it means having to reorder the sweeps inside of the driver. But, the issue will be how to handle the data retrival at the routine. We can try to take that into account:

- in the driver itself which can be a little tricky but doable
- setting a convention for the frequency being the first sweep
- communicating this change to back to the routine in some way

I would love to hear from the other drivers @aorgazf, @stavros11, @rodolfocarobene if that kind of limitation appears on your device and how they handle the averaging on 2D sweeps.

All this also made me notice how 2D sweeps may be different in some cases were changing its order may lead to different results due to how we are averiging the noise when some of the parameters may suffer from noise in the time scales of the experiment.

andrea-pasquale, 2023-05-02:
> Since we are free to select the sweep order in qibocal, I think the easiest would be to reorder the sweeps in all 2D routines, so that they are supported by all drivers. For example if Zurich (and maybe qick has a similar limitation) cannot sweep frequencies as inner loop, then switch them to outer loop in qibocal. It will still work with QM. I am not sure about qblox.

We could also think about having a custom objects to collect the output of `platform.sweep` at this point.

Jacfomg, 2023-05-04:
So, after thinking about it even with shoots as the outer loop the order on the parameter loops will mean that the middle one will be sequential in respect to the inner one. That would mean that it will not be the same to do exchange them but I am also unsure of how this can affect as the timescale of the noise affecting the parameters and the time of the inner loops take will vary with the experiment. I would suggest to set the convention for future routines to use inner loops for the most sensitive parameters, like flux.

closed 2023-07-05 by Jacfomg