qiboteam/qibolab #403: Using QCodes as the base for instruments

maxhant, 2023-05-03:
To avoid having to write wrappers for all instruments' drivers out there, it would be nice if our `AbstractInstrument` was inheriting the `Instrument` class of `Qcodes`. It would be easier for labs to join the project.

Currently, I am not sure what value it brings to have the wrapper. 
The `start` `setup` `connect` could just be in the `__init__`, while the `close` `disconnect` `stop` could be in just `close` as it is done in Qcodes: https://qcodes.github.io/Qcodes/api/instrument/index.html#qcodes.instrument.Instrument.close

I would like to start this discussion here!

Jacfomg, 2023-05-08:
At least or Zurich they are way too outdated using some old devices and their past libraries. So, they would not be an option for them, maybe for other devices they are better.

maxhant, 2023-05-10:
> At least or Zurich they are way too outdated using some old devices and their past libraries. So, they would not be an option for them, maybe for other devices they are better.

This is not a problem because you just add it then in the instruments of qibolab, and you just inherit from the QCodes instrument base.