qiboteam/qibolab #405: Change units to s on the drivers

Jacfomg, 2023-05-08:
I find it clearer to having runcard working with seconds for relaxation times

stavros11, 2023-05-08:
I think the reason we are using ns is that most instruments (maybe at least QM and qblox?) use that and we are also sure that we can use `int`, while for seconds we would need floats and most like `1e-6` notation. Personally, I am fine with any choice.

Jacfomg, 2023-05-09:
I just tried a runnning half nanosecond pulses directly on the driver and it didn't seem to complain. Also I have this from them

[image]

Jacfomg, 2023-11-14:
#660 we will have floats so that would use the full device resolution.