qiboteam/qibolab #423: Cannot install `tiiq` extras using pip

andrea-pasquale, 2023-05-18:
It seems that now it is not possible to install the lab dependencies (`tiiq`) through pip.
```sh
WARNING: qibolab 0.0.3 does not provide the extra 'tiiq'
``` 
I think that this is the case since all the instruments dependencies are in a poetry group
https://github.com/qiboteam/qibolab/blob/59a7748746207d4cc3c65950cb99f948e2e308af/pyproject.toml#L60
according to the poetry [documentation](https://python-poetry.org/docs/master/managing-dependencies/#dependency-groups) _groups can only be installed through poetry_.
I feel like we should convert it to an extra instead of a group to provide to the user the possibility of installing such deps using pip given that poetry should only be used by developers.
Thoughts @Edoardo-Pedicillo @AleCandido @scarrazza

AleCandido, 2023-05-18:
Indeed, I thought I told you in multiple places, but yes, that's exact.

The idea is that the dependencies that you're using for testing, e.g., are completely irrelevant to the user, since he does not need to run tests, but they are relevant to the developer (even though not necessarily always).

Instead, optional dependencies that are also relevant to the user should be collected in `pip` extras.

(You might wonder why there isn't a single dev group, but you might also remember that there was, and in principle we could stick to a single group. However, you do not need always all the dev dependencies, e.g. rtd only needs the docs deps.)

andrea-pasquale, 2023-05-18:
> Instead, optional dependencies that are also relevant to the user should be collected in `pip` extras.

I think that this is the case at least for `tiiq`

@Edoardo-Pedicillo could you have a look when you have time?

andrea-pasquale, 2023-05-22:
After the meeting today we agreed on introducing several extras based on the electronics (`qblox`, `zurich`, `qm`, ...)