qiboteam/qibolab #427: Implement sequence unrolling

Jacfomg, 2023-05-18:
We should consider this as a thing in the abstract platform level as @stavros11 suggested because it seems to work in Zurich as well. I'm still missing benchmarking on how many pulses, measurements I can make on each PulseSequence. It crashed at 1320 drive pulses of 40ns and 20 measurement ones of 2000ns. This is still enough from running most things way faster in Zurich because of the compilation steps.

The thing is that I believe we might need to calculate this time as a way to know how much memory in the device we will use to rearrange our PulseSequences accordingly before execution in bigger or smaller packages. Relaxation time applied on the inner sequences may not have a big impact if set properly as a delay from what I tried. But it would be great to know how qm or qblox behaves if you implemented it on both.

AleCandido, 2023-09-21:
This has been attempted in #465, but it should be resumed (whenever possible).

stavros11, 2023-09-21:
@AleCandido @Jacfomg I have a branch (`unrolling`) that implements a general unrolling at the level of `List[PulseSequence] -> PulseSequence]`. I am hoping that it will work out-of-the-box (without driver modifications) for at least QM and qblox, but I did not test it yet so I did not open a PR. I am planning to try next week. The idea is that `platform.execute_pulse_sequences` will first try to delegate the unrolling to the controller (via `instrument.play_sequences`) and if this is `NotImplemented` it will fall back to my general implementation.

AleCandido, 2023-09-21:
This is what I expected, but maybe it is worth to take a look at the PR to see if this is sufficient for Zurich.
Or just test on Zurich as well, and we will worry only if it will fail.