qiboteam/qibolab #429: Refactor AbstractPlatform

scarrazza, 2023-05-22:
Remove `AbstractPlatform` and propose a new `Platform` class.

AleCandido, 2023-05-30:
About the last part, there is no problem.
Also, the two things do not exclude each other: `ChannelMap` was just a tool to reduce duplication during creation, and it can be even more useful the more attributes you want to add:
```py
channels = ChannelMap(channel_type=QbloxChannel)
channels.current_instrument = qcm_rf1
channels |= ((f"L4-{n}", f"o{n}") for n in range(...))
channels.current_instrument = qcm_rf2
channels |= ((f"L5-{n}", f"o{n + ...}") for n in range(...))
```

Giving up on the `ChannelMap` implies one of three possible outcomes:
- reintroduce a lot of duplication, with a language that does not scale (because the iteration happens at the level of the source)
- repeat patterns among different `create_` functions (iteration happens dynamically, but with no common helper)
- create a new API in Qibolab that will have the same role of `ChannelMap`

AleCandido, 2023-05-30:
About the channel in the `qubit` there are many possible solutions. If you like that in particular, I would not object, since it would be a pointless discussion.
The main difference among the various layouts would be in the *ownership* of objects, and how they reference each other. This might cause unexpected behavior at a later stage (more or less as in C you have the feelings that you could do whatever with pointers, until you actually get the issues), but since it is not so intuitive, better to actually wait for the time when we'll have problems, and then it will much easier to understand why.

Just to cite another simple option, you could pass around some `PlatformQubit`:
```py
@dataclass
class PlatformQubit:
    qubit: Qubit
    platform: Platform

    def __getattribute__(self, name):
        try:
            return getattr(self.qubit, name)
        except AttributeError:
            return self.__getattr__(name)

    ...
```
In that case the `Qubit` can keep knowing nothing about the platform, but the `PlatformQubit` knows (since it is essentially a `Qubit` + a reference to a `Platform`).
However, I'm not even advocating for this, especially when you want to modify the `Platform` (because having multiple reference around, able to mutate the object, might create other issues).

AleCandido, 2023-05-30:
The thing about which I'm more opinionated is the `Channel` specialization: I don't see much of the point, when we already have `Instrument`s.

In principle, you want to have a common interface as soon as possible, to avoid dealing with duplicated code.
```
A1 -> B1 - \
A2 -> B2 - - > C
A3 -> B3 - /
```
is strictly worse than
```
A1 - \
A2 - - > B -> C
A3 - /
```
because not only you have to deal with the growing complexity of increasing `A`s, and possibly refactoring all of them, but even `B`s.

So, what is the advantage of creating another specialized object, instead of implementing a common API at the level of the `Instrument`s?

AleCandido, 2023-06-23:
This should have been addressed by #438 and #469. Can we close? @scarrazza @stavros11

closed 2023-06-23 by scarrazza