qiboteam/qibolab #435: Choose different classification methods

Edoardo-Pedicillo, 2023-05-23:
After implementing the training of different classification methods in `qibocal`, we could give the possibility to decide which one to use in the circuit execution.

AleCandido, 2023-09-26:
@Edoardo-Pedicillo is there any reason to keep pushing for this?

At this point, I believe that we're always using a single method, for the time being good enough.
Do you have any plan of using anything else?

andrea-pasquale, 2023-09-26:
Actually this will very relevant shortly.
In order to calibrate two qubit gates we will need to perform the classification for 3 states 0, 1 and 2. The current choice of fit is able only to address the classification with 0 and 1. Therefore it will be super useful to plug different classification methods.

AleCandido, 2023-09-26:
Ok, great to hear someone will take care :)

AleCandido, 2023-12-02:
@andrea-pasquale maybe we should reconsider this once more, and make an issue in Qibo itself for broad support of qutrits (whenever we'll have time for it, but not now).