qiboteam/qibolab #451: Zurich flux sweeper

AleCandido, 2023-05-26:
@stavros11 Zurich sweeps a flux pulse that is different than the offset applied at one port. Unfortunately, it is not a 1:1 match, so correcting a sweetspot is not direct. However, the offset parameter has a greater range, so it is advantageous. It is always on as well, which could be a problem. 
I think that this can be raised as an issue, and it does not prevent the merger.

_Originally posted by @maxhant in https://github.com/qiboteam/qibolab/pull/222#discussion_r1203963409_

Jacfomg, 2023-05-26:
Its slower and there should be a simple relation between them. We know the max Vpp of the HDAWG pulses and the amplitude of the pulse so there should be a relation between them that is not complicated. Just need to confirm it with oscilloscope before people complaining about this and get a worse sweeper.

Jacfomg, 2023-05-26:
The always on is a different thing and I already added some lines for that last week that I asked him to test. As he is afraid of pulling I don't even know if they worked or not.

maxhant, 2023-05-31:
> The always on is a different thing and I already added some lines for that last week that I asked him to test. As he is afraid of pulling I don't even know if they worked or not.

There isno PR open on this, and my branch is ahead on `main` just for the relative_phase and amplitude sweeper fixed that you introduced.
Where is the code that fixes it, and I will merge it?

Jacfomg, 2023-05-31:
As there relation I proposed was linear going from amplitude to voltage is actually direct and Maxime checked it on the scope and it holds I will not change the driver.

closed 2023-09-21 by Jacfomg