qiboteam/qibolab #454: Simplify dummy and remove paths.py

AleCandido, 2023-05-29:
The only thing that is actually being used is `qibolab_folder`, and mainly for retrieving runcards.

After #438, we do not have runcards in Qibolab any longer, apart from `dummy.yml`.

But the dummy runcard could be extremely simplified, and turned directly into a `dict` in a Python file, such that we won't need to load YAML any longer or locate it in the package.

At that point, we'll be able to fully drop `paths.py`.

scarrazza, 2023-05-29:
I fully agree, there is no point of storing a file which is not used by the qibolab API.

stavros11, 2023-05-29:
Regarding the dummy, at least for the time being (before we move to a database) there is some value in keeping the yaml, due to https://github.com/qiboteam/qibocal/pull/365#discussion_r1209059125. I would consider moving dummy to the qibolab_platforms_qrc repository so that we can re-enable qibocal tests that use it. @AleCandido let me know if you agree. This will also allow us to remove `paths.py`, although I believe some other paths are also currently somewhere in the code (but on things that are not very useful).

Note that the part of qibocal's `ActionBuilder` that copies the yaml runcard will need to be modified when we move to a database, so the above solution is temporary.

AleCandido, 2023-05-29:
I'm actively working on the database, but I'd be glad if we can have a solution even before.

On the other hand, it would be nice to have unit tests working without `qibolab_platforms_qrc` and without the database as well.
We can add a switch to hijack for dummy only.

stavros11, 2023-05-29:
> On the other hand, it would be nice to have unit tests working without `qibolab_platforms_qrc`, and without the database as well. We can add a switch to hijack for dummy only.

For qibolab tests we would still need to add the platform to `tests/dummy_qrc` folder. For qibocal tests, I am not sure, maybe @andrea-pasquale has a better idea. Right now the `ActionBuilder` is copying the runcard yaml and instantiates the platform using the copy, so that when it later dumps, it does not change the original runcard. So it needs to know where runcards are stored to perform the copy.

AleCandido, 2023-05-29:
Ok, I see, you want to move the runcard into `tests/`.

But the idea of keeping that single runcard (simplified as much as possible) in the package, rather than in the tests, was to give something a user could run out of the box (including the tests). Even though it does make no sense, but might be useful for development.

However, maybe for development people should copy the runcards from the `tests/dummy_qrc/`. But being one more step, I see the *embedded dummy* as a simplification for the user (but I would only put trivial parameters, like `0`, `0.`, `None`, `[]`, and similar).

AleCandido, 2023-05-29:
E.g.:

```py
# dummy.py
{
  "nqubits": 2,
  "description": "Dummy.",

  "qubits": [0, 1],

  "settings":
    {
      "sampling_rate": int(1e9),
      "relaxation_time": 0,
      "nshots": 1,
    },

  "resonator_type": "2D",
  ...
}
```

andrea-pasquale, 2023-05-29:
> For qibolab tests we would still need to add the platform to `tests/dummy_qrc` folder. For qibocal tests, I am not sure, maybe @andrea-pasquale has a better idea. Right now the `ActionBuilder` is [...]

Regarding qibocal I'm already able to test single routines using the `create_platform("dummy")`. In order to test the full pipeline (using `qq` or `qq-auto`), I would be also interested in testing the mechanism to update the runcard, which usually relies on the information written in the runcard.

> E.g.:
> 
> ```python
> # src/qibolab/dummy.py
> card = {
>   "nqubits": 2,
>   "description": "Dummy.",
> 
>   "qubits": [0, 1],
>   "topology": [],
> 
>   "settings":
>     {
>       "sampling_rate": int(1e9),
>       "relaxation_time": 0,
>       "nshots": 1,
>     },
> 
>   "resonator_type": "2D",
>   ...
> ```

Having something like this it should work I think. Moreover, I can also test the generation of the runcard using the method [dump](https://github.com/qiboteam/qibolab/blob/7a5b331498fe788620ac9ec3071fff71dcd745bc/src/qibolab/platform.py#L125) of the `AbstractPlatform`.

At the same time I feel like we should try to reduce the differences between the dummy and the other platforms, to make tests and debugging easier. Either solutions is fine by me. The sooner we implement it the sooner I can increase coverage in qibocal.

scarrazza, 2023-05-29:
Yes, let's go for a dict please.

andrea-pasquale, 2023-05-30:
Is someone working on this @AleCandido @stavros11 ?
If not I could give it a go, since I need this for qibocal.
As agreed I will implement the dict to see if this is enough for qibocal.

AleCandido, 2023-05-30:
> Is someone working on this @AleCandido @stavros11 ?
> If not I could give it a go, since I need this for qibocal.
> As agreed I will implement the dict to see if this is enough for qibocal.

If you make a PR, then we can discuss on that.

At the moment, I'm working on the DB, but I believe the dummy should stay outside of it anyhow.

closed 2023-05-31 by andrea-pasquale