qiboteam/qibolab #457: Outdated documentation

rodolfocarobene, 2023-05-29:
From the last documentation deployment, you can see various warning and errors related to outdated components (new results, abstrac platform...)

https://github.com/qiboteam/qibolab/actions/runs/5109472801/jobs/9184821662#step:6:210

We should fix them, but also we may want to fix the workflow so that it fails when these kind of errors present (?)

scarrazza, 2023-05-31:
We should update the README too.

scarrazza, 2023-06-29:
For the next release we need proper documentation with examples and detailed updated explanations. @rodolfocarobene @Edoardo-Pedicillo could you please take care of this (and integrate the new figures from @MatteoRobbiati)?

PR #458 (2023-05-31): Update doc
Fixes #457.

There are still some errors mainly related to unexpected indentations

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [ ] Tests are passing.
- [ ] Coverage does not decrease.
- [ ] Documentation is updated.
PR #546 (2023-08-08): Update documentation
Finally closes #457.

Checklist:
- [x] Automatic documentation (api-reference).
- [x] Examples/description for all the main components of qibolab.
- [x] Examples of higher level use (experiments/circuits).

The documentation can be compiled locally with `make -C doc html` or with `poe docs`. 
This is clearly still a draft :-)
closed 2023-08-08 by rodolfocarobene