qiboteam/qibolab #466: Qubit method to get channel from pulse type

rodolfocarobene, 2023-06-01:
Following a discussion with @AleCandido in https://github.com/qiboteam/qibolab/pull/373#discussion_r1212815266, we could add a function to the `Qubit` object for retrieving the channel object using a pulse type.

Something like:
```python
class Qubit:
  ...
  def get_channel_from_pulse_type(self, pulse_type: PulseType):
    channel_name = pulse.type.name.lower()  # would be 'readout', 'drive', 'flux'
    channel = getattr(self, pulse_type)
    return channel
```
To be used in the drivers with:
```python
qubit = qubits[pulse.qubit]
channel = qubit.get_channel_from_pulse_type(pulse.type)
```

Would this be useful to other drivers? (@aorgazf, @Jacfomg, @stavros11)

Jacfomg, 2023-06-01:
As far I managed to have it working without using channels at least for Zurich

AleCandido, 2023-06-01:
> As far I managed to have it working without using channels at least for Zurich

How did you get ports?

This is really a matter of coordination, we should have one favorite (and possible unique) way of doing things.

---

*Btw, I was thinking more to something like:*
```py
class Qubit:
  ...
  def __getitem__(self, pulse_type: PulseType):
    channel_name = pulse.type.name.lower()  # would be 'readout', 'drive', 'flux'
    return getattr(self, pulse_type)
```
*but they are both fine xD*

AleCandido, 2023-06-01:
Ok, so Zurich has currently its own way of dealing with ports.
We definitely need something with broader adoption, otherwise we'll have to maintain all the different options.

alecandido, 2024-07-16:
Given the current rework of qubits and channels in 0.2, this issue is just outdated.

The need of retrieving channels is well-known, and going to be addressed in #885 and future developments.

closed 2024-07-16 by alecandido