qiboteam/qibolab #470: Remove dependency on setuptools

rodolfocarobene, 2023-06-05:
Currently `qualang` depends on `dash` that has an undeclared dependency on `setuptools`, while waiting for a fix from `dash` (https://github.com/plotly/dash/issues/2557), we will add explicitly the dependency in `pyproject.toml`.

This issue is to remember us to remove it, when possible

AleCandido, 2023-09-26:
https://github.com/plotly/dash/commit/2bb4f124147a0a0269bf5064e72c347812c4443d

stavros11, 2023-09-28:
Not sure if it will fix this issue, but noting that I am also expecting newer releases for qm-qua and qualang_tools for fixing the py3.11 issue (#588). Hopefully we can resolve both then.

stavros11, 2023-11-07:
setuptools is now a requirement for the latest versions of dash so I removed it from our requirements in #644.

closed 2023-11-07 by stavros11