qiboteam/qibolab #478: Is Pycln working?

rodolfocarobene, 2023-06-09:
Is pycln working properly? It happened to me a couple of times that there unused imports that pycln did not detect, even when changing the file.

If someone wants to try, here `SweeperType` is unused:
https://github.com/qiboteam/qibolab/blob/66fc74aae37ac2fd2261bd882487ff797d57d1a2/src/qibolab/instruments/rfsoc.py#L23

Did someone else noticed it or it's just my installation?

AleCandido, 2023-06-12:
*side effects* are things running during import, other than definitions. Essentially, global statements.

However, this should happen at module level: removing one module import might avoid something running. But it shouldn't happen for individual variables inside a module.

andrea-pasquale, 2023-06-12:
I also think it should be okay.
We can give it a try and if we see something wrong we can drop it.

closed 2023-06-12 by rodolfocarobene