qiboteam/qibolab #494: Interface for sequence unrolling

Jacfomg, 2023-06-15:
Thanks a lot for implementing this!

Regarding the interface, I was thinking to have a different method (for example `instrument.play_sequences` and `platform.execute_pulse_sequences`) for when passing the list of sequences, in contrast to `platform.execute_pulse_sequence` for a single sequence. What you did is also fine, it is just a bit more complicated when the same function behaves differently depending on the input type. Overall, personally I don't have a strong preference, let's see what others think.

Regardless, it would be good to document that:
* if you accept lists on `play`, modify the docstring of `qibolab.instruments.abstract.Controller.play` to explain this.
* if a list is passed the result is a `Dict[List[ResultType]` instead of the usual `Dict[ResultType]`.

Also good to add a test that is passing a list of sequences to the dummy play.

_Originally posted by @stavros11 in https://github.com/qiboteam/qibolab/pull/493#pullrequestreview-1480318122_

Jacfomg, 2023-06-19:
We agreed on using `execute_pulse_sequences`.

Jacfomg, 2023-11-14:
@stavros11 this can be closed, right ?

stavros11, 2023-11-14:
I think so, we have #427 for unrolling anyway.

closed 2023-11-14 by Jacfomg