qiboteam/qibolab #505: Multidimensional time sweepers

AleCandido, 2023-06-26:
> I believe this would not work if `len(sweeper.pulses) > 1`, because you would append the start & stop of pulses that are after a couple of pulses being swept twice.

_Originally posted by @AleCandido in https://github.com/qiboteam/qibolab/pull/474#discussion_r1234224289_

alecandido, 2024-07-16:
This should not be a problem any longer in 0.2, since `starts` and `stops` do not need any longer to be computed, at least from Qibolab perspective.

Moreover, we have very limited support for duration sweepers on non-rectangular and non-delay duration sweepers. Thus, I wonder whether we still need to track this.

stavros11, 2024-09-06:
Indeed, since there’s no start and stop for pulses anymore, I believe we can close.

closed 2024-09-06 by stavros11