qiboteam/qibolab #516: EraSynth Connection issues

Jacfomg, 2023-07-05:
When doing RB or just in general they may fail to connect even if you increase to 10 the number of tries for connection. That happens around every hour at least once. During RB this connection needs to be established at every circuit execution so if it fails all data is lost and the experiment execution time is limited by when the LO will fail. We can either try to save data or find a way to deal with the EraSynth.

stavros11, 2023-07-05:
We could also establish the connection only once in the beginning and disconnect in the end. Reestablishing connection after every execution sounds suboptimal regardless of EraSynth issues. In fact looking at `QibolabBackend.execute_circuit` I am not sure why it is trying to connect every time, it should only connect the first time and then skip this step since it is already connected. Maybe there is another bug somewhere? Saving data periodically if you have that option is also a good strategy against failures.

That being said, we still need to deal with the EraSynth connection issues regardless of the above.

Jacfomg, 2023-07-05:
I reproduced the error a couple times and all of them was something like this: 

`requests.exceptions.ConnectTimeout: HTTPConnectionPool(host="TWPA_pump_lo_ip", port = 80): Max retries exceeded with url ` ...

after executing circuits for about 1 hour. So it seems like something is trying to reconnect to the EraSynth for some reason.

Jacfomg, 2023-07-06:
Try:

time.delay for reconnection bigger,
platform.stop() on the circuit execution comment
Just comment the TWPa on the create and turn it on and off by hand.

PR #529 (2023-07-11): Fix Erasynth issues
Should fix #516. I've followed some answers [here](https://stackoverflow.com/questions/28377421/why-do-i-receive-a-timeout-error-from-pythons-requests-module). 
For me the problem seems solved/much less relevant, but could you also try this please, @Jacfomg?

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [ ] Tests are passing.
- [ ] Coverage does not decrease.
- [ ] Documentation is updated.
closed 2023-07-11 by rodolfocarobene