qiboteam/qibolab #531: Tests cleanup

aorgazf, 2023-07-12:
Hi @stavros11,
As discussed, with the removal of Multiqubit and the change of interface to control instruments (from the platform get/set methods to ports) there is a need for cleaning up the tests.
Could you please review and clean:
conftest.py
test_platforms_abstract.py
test_platforms_design.py
test_instruments_erasynth.py
test_instruments_rohde_schwarz.py
test_instruments_qutech.py

and also try to fix the execution of tests on qpu initiated from github.

Please work on my branch 
alvaro/autocalibration_qblox9_sweepers_refactor_ports

aorgazf, 2023-07-14:
hi @stavros11,
could you have a look at test_backends.
when running on qpu it throws `Hardware backend only supports circuits as initial states.`

aorgazf, 2023-07-14:
hi @stavros11,
In `test_platforms_abstract.py` there are a number of tests using the fixture `platform`.
I noticed the platform is loaded for every test, so it takes a long time.
This fixture currently has a function scope. i tried to change its scope to module, but it complains that it is using the fixture platform_name, which has a function scope.
could you please have a look into it.

Could you also look into fixing the failing tests?
thanks

aorgazf, 2023-07-14:
Hi @rodolfocarobene,
A few qpu tests in test_instruments_rfsoc.py are failing:
```
test_instruments_rfsoc.py::test_call_executepulsesequence FAILED         [  5%]
test_instruments_rfsoc.py::test_call_execute_sweeps FAILED               [  5%]
test_instruments_rfsoc.py::test_play_qpu FAILED                          [  5%]
test_instruments_rfsoc.py::test_sweep_qpu FAILED                         [  5%]
test_instruments_rfsoc.py::test_python_reqursive_sweep FAILED    
```

aorgazf, 2023-07-14:
Hi @stavros11,
I've run all tests in qpu. below is the list of failed ones:
```

=========================== short test summary info ============================
FAILED test_backends.py::test_execute_circuit_initial_state[qm] - TypeError: ...
FAILED test_backends.py::test_execute_circuit[qm-I-kwargs0] - ValueError: Har...
FAILED test_backends.py::test_execute_circuit[qm-X-kwargs1] - ValueError: Har...
FAILED test_backends.py::test_execute_circuit[qm-Y-kwargs2] - ValueError: Har...
FAILED test_backends.py::test_execute_circuit[qm-Z-kwargs3] - ValueError: Har...
FAILED test_backends.py::test_execute_circuit[qm-RX-kwargs4] - ValueError: Ha...
FAILED test_backends.py::test_execute_circuit[qm-RY-kwargs5] - ValueError: Ha...
FAILED test_backends.py::test_execute_circuit[qm-RZ-kwargs6] - ValueError: Ha...
FAILED test_backends.py::test_execute_circuit[qm-U3-kwargs7] - ValueError: Ha...
FAILED test_backends.py::test_measurement_samples[qm] - ValueError: Hardware ...
FAILED test_backends.py::test_ground_state_probabilities_circuit[qm] - ValueE...
FAILED test_backends.py::test_excited_state_probabilities_circuit[qm] - Value...
FAILED test_backends.py::test_superposition_for_all_qubits[qm] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[qblox-I-kwargs0] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[qblox-X-kwargs1] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[qblox-Y-kwargs2] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[qblox-Z-kwargs3] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[qblox-RX-kwargs4] - ValueError:...
FAILED test_backends.py::test_execute_circuit[qblox-RY-kwargs5] - ValueError:...
FAILED test_backends.py::test_execute_circuit[qblox-RZ-kwargs6] - ValueError:...
FAILED test_backends.py::test_execute_circuit[qblox-U3-kwargs7] - ValueError:...
FAILED test_backends.py::test_measurement_samples[qblox] - ValueError: Hardwa...
FAILED test_backends.py::test_ground_state_probabilities_circuit[qblox] - Val...
FAILED test_backends.py::test_excited_state_probabilities_circuit[qblox] - Va...
FAILED test_backends.py::test_superposition_for_all_qubits[qblox] - ValueErro...
FAILED test_backends.py::test_execute_circuit_initial_state[rfsoc] - Connecti...
FAILED test_backends.py::test_execute_circuit[rfsoc-I-kwargs0] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[rfsoc-X-kwargs1] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[rfsoc-Y-kwargs2] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[rfsoc-Z-kwargs3] - ValueError: ...
FAILED test_backends.py::test_execute_circuit[rfsoc-RX-kwargs4] - ValueError:...
FAILED test_backends.py::test_execute_circuit[rfsoc-RY-kwargs5] - ValueError:...
FAILED test_backends.py::test_execute_circuit[rfsoc-RZ-kwargs6] - ValueError:...
FAILED test_backends.py::test_execute_circuit[rfsoc-U3-kwargs7] - ValueError:...
FAILED test_backends.py::test_measurement_samples[rfsoc] - ValueError: Hardwa...
FAILED test_backends.py::test_ground_state_probabilities_circuit[rfsoc] - Val...
FAILED test_backends.py::test_excited_state_probabilities_circuit[rfsoc] - Va...
FAILED test_backends.py::test_superposition_for_all_qubits[rfsoc] - ValueErro...
FAILED test_instruments_rfsoc.py::test_call_executepulsesequence - TypeError:...
FAILED test_instruments_rfsoc.py::test_call_execute_sweeps - TypeError: _exec...
FAILED test_instruments_rfsoc.py::test_play_qpu - ConnectionRefusedError: [Er...
FAILED test_instruments_rfsoc.py::test_sweep_qpu - ConnectionRefusedError: [E...
FAILED test_instruments_rfsoc.py::test_python_reqursive_sweep - ConnectionRef...
FAILED test_instruments_zhinst.py::test_experiment_execute_pulse_sequence - V...
FAILED test_instruments_zhinst.py::test_experiment_sweep_2d_specific - ValueE...
FAILED test_instruments_zhinst.py::test_connections - ValueError: Platform iq...
FAILED test_platforms_abstract.py::test_platform_execute_empty[rfsoc] - Conne...
FAILED test_platforms_abstract.py::test_platform_execute_one_drive_pulse[qm]
FAILED test_platforms_abstract.py::test_platform_execute_one_drive_pulse[qblox]
FAILED test_platforms_abstract.py::test_platform_execute_one_drive_pulse[rfsoc]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_one_drive_one_readout[qm]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_one_drive_one_readout[qblox]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_one_drive_one_readout[rfsoc]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_multiple_drive_pulses_one_readout[qm]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_multiple_drive_pulses_one_readout[qblox]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_multiple_drive_pulses_one_readout[rfsoc]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_multiple_drive_pulses_one_readout_no_spacing[qm]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_multiple_drive_pulses_one_readout_no_spacing[qblox]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_multiple_drive_pulses_one_readout_no_spacing[rfsoc]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_multiple_overlaping_drive_pulses_one_readout[qm]
FAILED test_platforms_abstract.py::test_multiqubitplatform_execute_multiple_overlaping_drive_pulses_one_readout[qblox]
[... cut ...]
ERROR test_platforms_abstract.py::test_ground_state_probabilities_pulses[qm-True]
ERROR test_platforms_abstract.py::test_ground_state_probabilities_pulses[qblox-False]
ERROR test_platforms_abstract.py::test_ground_state_probabilities_pulses[qblox-True]
ERROR test_platforms_abstract.py::test_ground_state_probabilities_pulses[zurich-False]
ERROR test_platforms_abstract.py::test_ground_state_probabilities_pulses[zurich-True]
= 74 failed, 2268 passed, 8 skipped, 8 warnings, 32 errors in 296.17s (0:04:56) =
```

rodolfocarobene, 2023-07-14:
I updated the tests for the rfsoc in #532, if they are executed locally they work and pass.

However in the selfhosted they wont since the platform called in "create_platform" does not exist...
I suppose we can fix it but we should re-think a bit the qpu tests as we were saying yesterday. Maybe this can be done in a second moment (?)

closed 2023-08-18 by aorgazf