qiboteam/qibolab #560: Improve qibolab.utils

stavros11, 2023-08-08:
For the time being we could even keep as it is, to make a smoother transition, but eventually I'd prefer to make more isolated functions, that would not depend on the whole runcard and its structure, but just on the relevant section.

_Originally posted by @AleCandido in https://github.com/qiboteam/qibolab/pull/525#discussion_r1285561735_

Also `Platform.dump` could be moved to `qibolab.utils` as it is the last piece of the `Platform` that is based on the TII runcard format.

stavros11, 2023-08-08:
Indeed, you are right that everything in `qibolab.utils` is for loading/dumping runcards. @scarrazza @rodolfocarobene do you have a preference for the name? 
Renaming is fairly simple and maybe it is even better to do it before the release so that we don't break the interface again after and confuse people.

scarrazza, 2023-08-08:
I agree. I don't like `.runcard` because it locks the concept of a future database and `.serialize` is too specific.

`.load` and `.io` are not bad, both are generic to accommodate future changes.

AleCandido, 2023-08-08:
On the one side, I was not incredibly convinced of `load`, because it also dumps. But on the other `io` could have been too generic, because I wonder if we'll ever have further IO other than serialization (like inputs and outputs for programs in a more structured way).

However, in practice they are both fine, just decide your favorite.