qiboteam/qibolab #570: Upgrade Qblox Drivers

aorgazf, 2023-08-18:
Hi,
qblox has released a new version of their firmware https://gitlab.com/qblox/releases/cluster_releases/-/releases
It would be nice to upgrade the current implementation of qblox driver to support this new version and take advantage of the features introduced.

andrea-pasquale, 2023-09-18:
@PiergiorgioButtarini @aorgazf this issue can be closed, right?

andrea-pasquale, 2023-09-18:
Actually nevermind I thought that #580 already closed this instead we still need to upgrade the qblox driver

PiergiorgioButtarini, 2023-09-18:
I think it can be closed, the driver are updated and all the cluster now have the new firmware that works with qblox-instruments v0.11.0. 
There is an additional feature that was added in the new firmware but since is not relevant for us I didn't implemented, so I think this issue can be closed.

andrea-pasquale, 2023-09-18:
Great, thanks! I will close it now.

closed 2023-09-18 by andrea-pasquale