qiboteam/qibolab #576: Expose TWPA parameters in runcard

andrea-pasquale, 2023-08-31:
In order to be able to perform an automatic update of the platform when running protocols such https://github.com/qiboteam/qibocal/pull/501 we need to set some parameters which currently are hardcoded in the create_function. See for example here:

https://github.com/qiboteam/qibolab_platforms_qrc/blob/ad001f93f0b5eb2f39c632cd260a9ee644f9adb5/qw5q_gold_qblox.py#L145

The easiest way will be to put those values inside the runcard in order to perform the update.

If nobody argues with this I will adopt this approach.

Let me know if you agree @scarrazza @AleCandido @stavros11 @aorgazf

aorgazf, 2023-09-01:
Hi @andrea-pasquale,
In the last main refactor in qibolab, we replaced the platform runcard that contained both instrument settings and QPU properties, with a runcard for the QPU and a `create` method with hardcoded settings.

This gave the user flexibility on how to code the `create` method, but since it no longer has a predefined structure, we are now faced with the problem of not being able to automatically update those hard-coded parameters.

I do not like very much the idea of including instrument parameters to the QPU runcard. I would prefer to keep all instrument settings in an instruments runcard. Perhaps those settings can be loaded by the `create` method.

AleCandido, 2023-09-01:
In general, I like the idea of having instruments separate from QPU, because they are really two physically different objects.

For sure, while the lines' definition is perfectly suitable for the `create` method, and there is no need to change it (unless there is the need of some rework in the lab, that can be reflected by a `create` update), the parameters should not be stuck forever, requiring a function update.
About this we all agree.

I'm not sure about which is the best course of action in the transient phase: on the one side, I'd be tempted to suggest to go closer the proper layout (what Stefano is postponing to the *until* part), but on the other this would be a breaking change for the published Qibolab.

In any case, if in the QPU, it should be in a clearly separated section (e.g. `instruments`) such that it will be as simple as possible to take out of there once we'll decide it is time to do it.

scarrazza, 2023-09-01:
@AleCandido indeed a separate section should be sufficient. Managing more than one file with calibration and setup values might become cumbersome at the moment (until we have a deployment in the cluster).

PR #585 (2023-09-26): Instruments section in runcard
Fix #576. Needed for qibocal routines that update instrument (not qubit) parameters, such as the TWPA calibration. It is also useful to dump some instrument settings (LO frequency, etc.) in the runcard so that we can keep them in record. qibolab_platforms_qrc also needs to be updated for this to work.

@rodolfocarobene can you please confirm that my change in
https://github.com/qiboteam/qibolab/blob/f896cc2dd13890dc40e758281ce05bb182f88fc5/src/qibolab/instruments/rfsoc/convert.py#L39

is okay?

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [ ]  [... cut]
closed 2023-09-26 by stavros11