qiboteam/qibolab #584: 'Pulses execution' tutorial is not updated to work with Qibolab v0.1.0

jykhoo1987, 2023-09-20:
https://qibo.science/qibolab/stable/tutorials/pulses.html

The code that appears there cannot be run as it is. 

Fixes required for 1st code block:
1) 'phase' in the declaration of pulses should instead be 'relative_phase'
2) 'channel' keyword argument needs to be declared
3) suggestion to also declare 'qubit' keyword argument to make that option transparent to users

Fix required for 2nd code block:
In line 'results = platform.execute_pulse_sequence(ps, options=options)', 'ps' here should be instead 'sequence' if it is meant to be compatible with 1st code block.

Additional issue with pulse sequences:
'sequence' that is constructed in 1st code block has issues with sequence.plot(). It returns the error attached at the end of this issue. One way to bypass this error is to avoid the exact value of 60 for duration when declaring pulses. In this specific example in the 1st code block, 'duration=60' used for declaring the DrivePulse is causing this error. sequence.plot() will run correctly by changing it to any other value. Likewise, if we subsequently set 'duration' to 60 for the ReadoutPulse in this example, the error will resurface when running sequence.plot().

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [4], in <cell line: 1>()
----> 1 sequence.plot()

File ~/QOC/qibolabv_0_1_0/qibolab/src/qibolab/pulses.py:1981, in PulseSequence.plot(self, savefig_filename)
   1979 num_samples = int(pulse.duration / 1e9 * PulseShape.SAMPLING_RATE)
   1980 time = pulse.start + np.arange(num_samples) / PulseShape.SAMPLING_RATE * 1e9
-> 1981 ax.plot(
   1982     time,
   1983     pulse.shape.modulated_waveform_q.data,
   1984     c="lightgrey",
   1985 )
   1986 ax.plot(
   1987     time,
   1988     pulse.shape.modulated_waveform_i.data,
   1989     c=f"C{str(n)}",
   1990 )
   1991 ax.plot(
   1992     time,
   1993     pulse.shape.envelope_waveform_i.data,
   1994     c=f"C{str(n)}",
   1995 )

File ~/anaconda3/envs/QiboLabv0_1_0/lib/python3.9/site-packages/matplotlib/axes/_axes.py:1721, in Axes.plot(self, scalex, scaley, data, *args, **kwargs)
   1478 """
   1479 Plot y versus x as lines and/or markers.
   1480 
   (...)
   1718 (``'green'``) or hex strings (``'#008000'``).
   1719 """
   1720 kwargs = cbook.normalize_kwargs(kwargs, mlines.Line2D)
-> 1721 lines = [*self._get_lines(self, *args, data=data, **kwargs)]
   1722 for line in lines:
   1723     self.add_line(line)

File ~/anaconda3/envs/QiboLabv0_1_0/lib/python3.9/site-packages/matplotlib/axes/_base.py:303, in _process_plot_var_args.__call__(self, axes, data, *args, **kwargs)
    301     this += args[0],
    302     args = args[1:]
--> 303 yield from self._plot_args(
    304     axes, this, kwargs, ambiguous_fmt_datakey=ambiguous_fmt_datakey)

File ~/anaconda3/envs/QiboLabv0_1_0/lib/python3.9/site-packages/matplotlib/axes/_base.py:499, in _process_plot_var_args._plot_args(self, axes, tup, kwargs, return_kwargs, ambiguous_fmt_datakey)
    496     axes.yaxis.update_units(y)
    498 if x.shape[0] != y.shape[0]:
--> 499     raise ValueError(f"x and y must have same first dimension, but "
    500                      f"have shapes {x.shape} and {y.shape}")
    501 if x.ndim > 2 or y.ndim > 2:
    502     raise ValueError(f"x and y can be no greater than 2D, but have "
    503                      f"shapes {x.shape} and {y.shape}")

ValueError: x and y must have same first dimension, but have shapes (59,) and (60,)

stavros11, 2023-09-21:
Thank you @jykhoo1987 for reporting these issues. I pushed the fix for both in #587. If you can, please let us know if this works for you.

jykhoo1987, 2023-09-23:
Thank you @stavros11, it works now

PR #587 (2023-09-25): Fix pulse tutorial in docs and bug with sequence plotting
Fixes #584.

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [x] Tests are passing.
- [x] Coverage does not decrease.
- [x] Documentation is updated.
closed 2023-09-25 by stavros11