qiboteam/qibolab #593: create_CZ_pulse_sequence does not work with instructions in tutorial

jykhoo1987, 2023-09-23:
I am referring to this part under 'Tutorials' of the Qibolab v0.1.0 documentation: 'How to connect Qibolab to your lab?'
https://qibo.science/qibolab/stable/tutorials/lab.html

and in particular, the 2 code blocks at the bottom of the page.

The first is a runcard code block. I have copied the contents to a file 'my_platform.yml'.
The second is the platform construction code via the 'create()' function. I have copied the contents to a file 'my_platform.py'.

In a separate notebook, I am now trying to create an instance of this (dummy) runcard.

```python
from my_platform import create
from qibolab.backends import QibolabBackend

my_platform = create()
my_backend = QibolabBackend(platform=my_platform)
``` 

```python
my_platform.create_RX_pulse(0)
``` 
runs properly and returns:
DrivePulse(0, 40, 0.05682, 5_800_563_000, 0, Drag(5, 0.04), ch3, 1)

However, I run into issues with the CZ pulse:
```python
my_platform.create_CZ_pulse_sequence([0,1])
``` 

whiich returns:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Input In [4], in <cell line: 1>()
----> 1 my_platform.create_CZ_pulse_sequence([0,1])

File ~/QOC/qibolabv_0_1_0/qibolab/src/qibolab/platform.py:378, in Platform.create_CZ_pulse_sequence(self, qubits, start)
    373 if pair not in self.pairs or self.pairs[pair].native_gates.CZ is None:
    374     raise_error(
    375         ValueError,
    376         f"Calibration for CZ gate between qubits {qubits[0]} and {qubits[1]} not found.",
    377     )
--> 378 return self.pairs[pair].native_gates.CZ.sequence(start)

File ~/QOC/qibolabv_0_1_0/qibolab/src/qibolab/native.py:179, in NativeSequence.sequence(self, start)
    177 for pulse in self.pulses:
    178     if isinstance(pulse, NativePulse):
--> 179         sequence.add(pulse.pulse(start=start))
    180     else:
    181         virtual_z_phases[pulse.qubit.name] += pulse.phase

File ~/QOC/qibolabv_0_1_0/qibolab/src/qibolab/native.py:99, in NativePulse.pulse(self, start, relative_phase)
     83 """Construct the :class:`qibolab.pulses.Pulse` object implementing the gate.
     84 
     85 Args:
   (...)
     91     or :class:`qibolab.pulses.FluxPulse` with the pulse parameters of the gate.
     92 """
     93 if self.pulse_type is PulseType.FLUX:
     94     return FluxPulse(
     95         start + self.relative_start,
     96         self.duration,
     97         self.amplitude,
     98         self.shape,
---> 99         channel=self.qubit.flux.name,
    100         qubit=self.qubit.name,
    101     )
    103 pulse_cls = PulseConstructor[self.pulse_type.name].value
    104 channel = getattr(self.qubit, self.pulse_type.name.lower()).name

AttributeError: 'NoneType' object has no attribute 'name'

stavros11, 2023-09-23:
It is a bug in the platform `create`, because the CZ gate contains flux pulses, but we did not register flux channels to the qubits. I fixed it in f686f8ead355071e78cc9c1b9863b4a1176bef90 and pushed to #587.

Thank you for checking the examples in the docs. Unfortunately we did not have time to implement tests for them yet so it is likely there are bugs like that.

jykhoo1987, 2023-09-23:
Thank you for making the corrections.

closed 2023-09-25 by stavros11