qiboteam/qibolab #602: Add native pulse to generate $\ket{2}$ to platform

andrea-pasquale, 2023-09-27:
In order to efficiently calibrate two qubit gates we will need to be able to generate and classify also the state $\ket{2}$.
See for example qiboteam/qibocal#477 and qiboteam/qibocal#466.
I'm currently working on something in qibocal to calibrate this pulse.
I guess that the simplest solution would be to add also a native pulse for the this state, right?
Let me know what you think @stavros11 @AleCandido @igres26 @aorgazf @scarrazza

stavros11, 2023-09-27:
> I guess that the simplest solution would be to add also a native pulse for the this state, right? 

Since it is possible to calibrate such pulse, adding it as native sounds an acceptable solution to me.

> For a quick summary on how to calibrate this pulse you can have a look here https://learn.qiskit.org/course/quantum-hardware-pulses/accessing-higher-energy-states-with-qiskit-pulse

That is useful, thanks. I think we have been using another (but quite similar) way of finding the 0->2 frequency which is to do the qubit spectroscopy with high power starting from 0 and look for a peak that has lower frequency than the 0->1. This peak is at half the 0->2 frequency. But as far as I know we haven't tried Rabi oscillations on 1->2.

andrea-pasquale, 2023-09-27:
> That is useful, thanks. I think we have been using another (but quite similar) way of finding the 0->2 frequency which is to do the qubit spectroscopy with high power starting from 0 and look for a peak that has lower frequency than the 0->1. This peak is at half the 0->2 frequency.

Yes, indeed I believe that this what is described under "Computing the 1->2 frequency" bullet-point 1. Since once you have the $w_{02}$ you can also compute $w_{12}$ just by taking the difference with respect to $w_{01}$.

Here is a picture with the summary

[image]

We are planning to release qibolab at the of the week, right?
If so, I will try to open a PR before the end of the week with the new native pulse so that we will not need to release qibolab again in order to make it compatible with qibolab.

andrea-pasquale, 2023-09-27:
We will eventually need to store the parameter of the classifiers to classify $\ket{0}$,  $\ket{1}$ and  $\ket{2}$. But I believe that this will not be a problem. Let's try to do one at a time.

stavros11, 2023-09-27:
> We are planning to release qibolab at the of the week, right?

That was my plan, because we have already merged several important and breaking changes (qblox firmware, instrument settings, platform update, soon the couplers), but I guess we will decide that tomorrow. If there is something important to be added we can wait, but personally I do not mind releasing even every week, if that makes things more stable. The goal is to have the basic cluster environment (`module load qibo`) usable as much as possible, at least for basic calibrations and circuit executions.

PR #603 (2023-09-28): Adding `RX12` pulse
Closes #602.

Maybe the name for the pulse is not the best.
Let me know if you have a better idea.

@stavros11 concerning coverage should I just write a test for the new pulse?
Checklist:
- [ ] Reviewers confirm new code works as expected.
- [x] Tests are passing.
- [x] Coverage does not decrease.
- [x] Documentation is updated.
closed 2023-09-28 by scarrazza