qiboteam/qibolab #605: Use `Qubit` objects in pulses

stavros11, 2023-09-28:
Currently `pulse.qubit` is the qubit name of type str/int. It could be useful to instead use the `Qubit` objects. For example this would help distinguish qubits from couplers and get rid of the `CouplerFluxPulse` (https://github.com/qiboteam/qibolab/pull/508#discussion_r1339622652).

andrea-pasquale, 2023-09-28:
Just a remainder that whenever this is implemented also qibocal will need to change accordingly, given that we access several times the `qubit` attribute in pulses.

stavros11, 2023-09-28:
> Just a remainder that whenever this is implemented also qibocal will need to change accordingly, given that we access several times the `qubit` attribute in pulses.

Indeed, this is why this is a major change and we should be sure that it is for the better before we do that. We should check all the places where `pulse.qubit` is used in qibolab and qibocal and if it is more convenient to have the `Qubit` directly we should go for it.

alecandido, 2024-08-08:
Pulses do not contain qubits any longer (in 0.2), and they are not going to contain them anymore.

Despite meaningful, this issue is definitely outdated. I'm closing.

closed 2024-08-08 by alecandido