qiboteam/qibolab #606: Transpiler and compiler wishlist

stavros11, 2023-10-03:
Here is a list of enhancements that I would like to have in the transpiler and compiler in the next months. Note that these are only for usability, not the algorithms used. For algorithms see #440.
Ordered in terms of priority:

- [ ] **[Transpiler/Compiler]** Make sure that the returned results correspond to the original circuit. The transpiler may internally reroute the qubits and measurements in order to map them to the hardware connectivity but we should make sure that what is returned to the user follows the circuit the user executed.
- [ ] **[Compiler]** Make sure circuits can be executed on platforms with qubits not named as (0, 1, 2, ...), which is the default qubit naming convention in qibo circuits. Particularly, we have platforms where qubits are named using strings. The transpiler works circuit to circuit so it should not be affected by this.
- [ ] **[Compiler]** Expanding the above, allow the user to select which hardware qubits to use. Again, I think we can implement this in a way so that the transpiler is not affected.
- [ ] **[Transpiler/Compiler]** Change the single qubit native gate from U3 to GPI2 (see https://github.com/qiboteam/qibolab/issues/360#issuecomment-1604308906)
- [ ] **[Transpiler]** Move all transpiler code to qibo. This is a good idea for several reasons: 1. it is circuit to circuit transformation so qibolab is not really involved, 2. may be useful for applications other than hardware execution, 3. there are people (@renatomello @AlejandroSopena ) who may be interested in this and are more involved in qibo than qibolab.
- [ ] **[Transpiler]** Potentially allow a mechanism to select different single qubit natives in the transpiler, similar to what we have for two qubit natives.
- [ ] **[Workflow]** Currently the transpilation happens every time the circuit is executed. This may slow down applications such as variational optimization. We could have a way to re-execute transpiled circuits, potentially after changing some parameters. This is not trivial to implement.

1-2 are essential. 3-4 are good to have, especially for benchmarking applications. 5 we can do immediately if everyone agrees. 6+ are enhancements that may take longer time.

Let me know if I forget something.

Simone-Bordoni, 2023-10-03:
As discussed with Stavros I will proceed in the following way:
1 add the basic Sabre transpiler to qibolab.
2 move transpiler to qibolab to qibo (issue 5).
3 match the output mapping with measurements to fix the first issue.
4 define priority of the other issues to solve them in order.

stavros11, 2023-11-13:
Moved to qiboteam/qibo#1085 and #657.

closed 2023-11-13 by stavros11