qiboteam/qibolab #643: Issue with 'is_satisfied' related to transpilers

jykhoo1987, 2023-11-01:
I think there are some inconsistencies with the update to qibolab v0.1.2 when switching to other transpilers. Following the tutorial in the documentation, I do

```python
default_CR_backend.transpiler = NativeGates(two_qubit_natives=NativeType.CZ)
``` 
and then run the circuit which is just an X gate:

```python
c = Circuit(2)
c.add(gates.X(0))
#c.add(gates.CZ(0,1)) # actually adding CNOT to emulator
c.add(gates.M(*range(2)))
``` 

```python
backend_job = default_CR_backend.execute_circuit(c, nshots=1000)
```

I get an error related to this 'is_satisfied' attribute, which after looking around some of the codes, I think was only made for the StarConnectivity class but was left out for the other transpiler options available:

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[14], line 2
      1 default_CR_backend.transpiler = NativeGates(two_qubit_natives=NativeType.CZ)
----> 2 backend_job = default_CR_backend.execute_circuit(c, nshots=1000)

File ~/QOC/qibolabv_0_1_2/qibolab/src/qibolab/backends.py:92, in QibolabBackend.execute_circuit(self, circuit, initial_state, nshots, fuse_one_qubit, check_transpiled)
     86 if initial_state is not None:
     87     raise_error(
     88         ValueError,
     89         "Hardware backend only supports circuits as initial states.",
     90     )
---> 92 if self.transpiler is None or self.transpiler.is_satisfied(circuit):
     93     native_circuit = circuit
     94 else:
     95     # Transform a circuit into proper connectivity and native gates

AttributeError: 'NativeGates' object has no attribute 'is_satisfied'
---------------------------------------------------------------------------

stavros11, 2023-11-01:
Thank you for reporting this. I can reproduce, however unfortunately we are at a stage of lifting the transpilers to the qibo repository (see #634), so I would avoid making changes in the transpiler code until this is done to avoid conflicts. Once the move is done, we will move your issue to qibo and address it.

In the meantime, an alternative would be
```py
from qibolab.transpilers import Passes, NativeGates

backend.transpiler = Passes([NativeGates(two_qubit_natives=NativeType.CZ)], connectivity=backend.platform.topology)
```
(for you `backend = default_CR_backend`), as I noticed that `Passes` (former `Pipeline`) still has the `is_satisfied` attribute. I noticed another bug here, because `Passes` has another `native_gates` attribute which does not follow what you input to `NativeGates` but this shouldn't affect the example you posted. Again we will address this in qibo.

Simone-Bordoni, 2023-11-03:
In future versions we will remove the is_satisfied with a general function (assert_transpiling). As star_connectivity was made before thinking a general pipeline it has many differences with the "standard" router step (it is actually a placer+router). But i think we will remove this transpiler when moving to qibo as it works only on a fixed connectivity unlike Sabre or ShortestPaths.

Simone-Bordoni, 2023-11-03:
For now i have just removed the check on .is_satisfied. this should solve the problem once #634 will be merged

jykhoo1987, 2023-11-05:
Thank you very much!

jykhoo1987, 2023-11-09:
For now, as a 'quick fix', I have commented out part of the backends.py script in my local repo as follows:

```python
        '''
        if self.transpiler is None or self.transpiler.is_satisfied(circuit):
            native_circuit = circuit
        else:
            # Transform a circuit into proper connectivity and native gates
            native_circuit, qubit_map = self.transpiler(circuit)
            # TODO: Use the qubit map to properly map measurements
            if check_transpiled:
                assert_transpiling(
                    native_circuit, self.transpiler.connectivity, self.transpiler.get_initial_layout, qubit_map
                )
        '''
        native_circuit = self.transpiler(circuit)
``` 

I will implement a more permanent solution when transpiler is moved from qibolab to qibo in the next version and it becomes clear how the transpiler will be called when setting up the backend.

stavros11, 2023-11-13:
Since the transpiler is already merged to qibo, I moved this to qiboteam/qibo#1086. @jykhoo1987 let us know if I forgot something there.

closed 2023-11-13 by stavros11