qiboteam/qibolab #647: Instruments return different types for the same execution options

stavros11, 2023-11-06:
Cause of qiboteam/qibocal#598:
```py
from qibolab import ExecutionParameters, create_platform, AveragingMode, AcquisitionType
from qibolab.pulses import PulseSequence

platform = create_platform("qw5q_gold_qblox")
#platform = create_platform("iqm5q")

sequence = PulseSequence()
sequence.add(platform.create_MZ_pulse(q, start=0))

platform.connect()
platform.setup()
platform.start()

options = ExecutionParameters(
    nshots=1000,
    acquisition_type=AcquisitionType.DISCRIMINATION,
    averaging_mode=AveragingMode.CYCLIC
)

results = platform.execute_pulse_sequence(sequence, options=options)

platform.stop()
platform.disconnect()

print(results[0])
print(results[0].probability())
print(type(results[0].probability()))
```
Zurich (`iqm5q`) gives:
```
<qibolab.result.AveragedSampleResults object at 0x7f0f01f00040>
[0.994]
<class 'numpy.ndarray'>
```
Qblox (`qw5q_gold`) gives:
```
<qibolab.result.AveragedSampleResults object at 0x7f43b11b0af0>
1.0
<class 'numpy.float64'>
```

Similar inconsistencies appear for other (acqusition type, averaging mode) pairs. We should decide what type is appropriate for each pair and have checks in all constructors of result.py to make sure that drivers return the proper types (and also qpu tests that check that these constructors do not fail).

For the above example, do we expect to have more than one value in `AveragedSampleResults`? If no, I would go with the qblox approach (return float) because 1. float is simpler than array, 2. qibocal now works with that and 3. `SampleResults.average -> AveragedSampleResults` currently gives a float both for Zurich and qblox. In that case we should fix Zurich and drop `AveragedSampleResults.__add__` because we cannot add elements to the float. If there are arguments to keep the array we should fix qblox.

@andrea-pasquale @DavidSarlle @Jacfomg let me know your preference.

Jacfomg, 2023-11-06:
What are the rfsoc and qm returning ? Also, I assume there can be instances where this can be useful, but I can't think of any right now.

andrea-pasquale, 2023-11-06:
Thanks @stavros11 for opening this issue.
I was planning to open an issue exactly like this. 

> For the above example, do we expect to have more than one value in `AveragedSampleResults`? If no, I would go with the qblox approach (return float) because 1. float is simpler than array, 2. [...]

For this particular case I believe that we should just return a float.
I think that it should also be possible to write tests for this type of checks, right?
Another possible solution for this could be to use [`Pydantic`](https://github.com/pydantic/pydantic).
In this way we can define precise type hints for each `*Results*` class in [qibolab.result](https://github.com/qiboteam/qibolab/blob/main/src/qibolab/result.py) and errors will be triggered directly in Python if the return type is not the type hint specified by the class. 
Perhaps this would require a bigger effort in the beginning but in the long run could be very beneficial for qibolab in general.

DavidSarlle, 2023-11-06:
@stavros11 thaks for opening this issue. To me, the aquisition type AveragedSampleResults, as the name indicates, should return an averaged result (float) so, does not make any sense to returna an array. 

Also, as you said, there are advantages if we return a float form the code perspective:

1. If I amnot wrong, QM and rfsoc are returning a numpy.float64
2. Qibocal is compatible with numpy.float64, so, less changes are required.
3. Float is simpler than array 

Regarding the test and the use of pydantic, I totally agree with @andrea-pasquale.

If we can not see/imagine any case where the array option could be useful, is clear to me what shoud be the choice.

stavros11, 2023-11-07:
Thank you for the response, I fully agree with pydantic, I will have a look at it. 

I think one instance when array is required is when using sweepers. So it may actually make sense to keep the arrays. Sorry I forgot to mention this in the first post.

I am not sure what QM and RFSoc are doing and unfortunately I cannot test because they are not available (QM is disconnected / RFSoc I am not sure but I don't have any platform). Anyway, the plan to close this is the following:
* check QM and RFSoc,
* convert qblox and anything else needed to arrays,
* add pydantic to the results,
* add QPU tests for all different (acquisition type, averaging mode) pairs to make sure there are no errors.

PR #690 (2023-12-06): Add tests for result shapes
Fixes #647 for qblox and adds qpu tests that check that the shapes are correct for each acquisition type (except raw) and averaging mode.

I would also like to fix the shapes for Zurich and also do a quick test with qibocal before merging.

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [ ] Tests are passing.
- [ ] Coverage does not decrease.
- [ ] Documentation is updated.
closed 2023-12-06 by stavros11