qiboteam/qibolab #650: Improve `Sweeper` interface

stavros11, 2023-11-07:
The current interface is a bit a limiting as it does not allow several things that are supported by some instrument, such as:
* time sweepers that affect multiple pulses (eg. in Rabi length, Ramsey routines) - I believe this is currently supported in a messy way in the drivers but is not clearly expressed when defining the `Sweeper`/`PulseSequence`,
* possibility to sweep multiple quantities on the same loop (not nested). Eg. this would be useful for qiboteam/qibocal#519.

alecandido, 2024-04-23:
This is either internal to the drivers (first point), or duplicated by #692 (second point).

I'd propose to close, as #692 is more detailed about the second point, and we don't need to track continuous drivers' improvement, unless there is a specific issue with them (but *supported in a messy way* is rather generic).

Could we close it @stavros11? Or do you prefer to keep it open?

stavros11, 2024-04-23:
> I'd propose to close, as #692 is more detailed about the second point, and we don't need to track continuous drivers' improvement, unless there is a specific issue with them (but _supported in a messy way_ is rather generic).

I agree regarding the second point. Regarding the first, I believe the "messy way" is fixed by switching to delay: see related comment and implementations in https://github.com/qiboteam/qibocal/pull/809/files#r1576191953. Rabi length and Ramsey are now cleaner, at least in terms of how the sweep is defined and also do not require "diagonal" sweepers.

I will close this.