qiboteam/qibolab #669: Expose OpenQASM API

scarrazza, 2023-11-28:
- create an openqasm API, i.e. control qibolab from string.
- expose the API from python and lower level languages (Rust, C, Fortran)

AleCandido, 2023-12-11:
> - create an openqasm API, i.e. control qibolab from string.

This we could even turn into a CLI, it will take little to nothing.

> - expose the API from python and lower level languages (Rust, C, Fortran)

Since it's pretty minimal, it won't take much (I hope), since I know how to call Python at least from Rust, PyO3 should also handle the dependencies for us.
https://pyo3.rs/v0.20.0/python_from_rust

However, I wonder how we should distribute it.

Uploading the crate to https://crates.io/ will take little, but that's only for Rust. And it won't be available with `pip install qibolab` (even more, you will need both).

An interesting, but involved, solution, would be to compile the library and distribute with the Python package, but:
- we have to make sure it's properly working, and this will take some time
- the `qibolab` package won't be pure Python any longer, and we will need to start compiling for the various archs

scarrazza, 2024-01-23:
Completed in #765 .

closed 2024-01-23 by scarrazza