qiboteam/qibolab #698: Devices memory limitations

AleCandido, 2023-12-05:
The #618 is a powerful tool, and it would be good enough for all applications, if it were not for memory limitations.

We do not expect performance differences between unrolling and sweepers (since they are both loading the pulses, and if possible the sweeper is the one requiring extra operations), and not even a memory difference in the measurement storage (they are making the same measurement).

Thus, the limitation is coming from the memory required to store the pulses to send.

However, they might depend on many factors, e.g. the type and length of pulses. Currently, the number of measurements is used to gauge the batches, e.g. in Zurich
https://github.com/qiboteam/qibolab/blob/ef111646565dd51eced74b51ab6ffaf91c125277/src/qibolab/instruments/zhinst.py#L1225-L1226
but this might be just an approximate metric.

So, we should properly understand which are the actual memory limitations, device by device.

alecandido, 2024-07-12:
Qblox is pretty explicit about some of them:

> ##### 1. Instruction Memory
> Memory storing the instructions of the Q1ASM program to be executed. Commands are either classical or real-time in nature, defining whether they are executed by the classical core or passed to the buffer to be executed by the real-time core. For QCM modules the maximum number of instructions per program is 16384 and for QRM modules the maximum number of instructions per program is 12288.
> ##### 2. Data Memory
> Data memory is a set of registers that are used to store data during the execution of a Q1ASM program. These registers are referenced as R# from 0 to 63.

https://qblox-qblox-instruments.readthedocs-hosted.com/en/main/cluster/q1_sequence_processor.html#instruction-memory

RoyStegeman, 2026-09-09:
We have a working (though not exhaustive/complete) implementation of taking into account memory limitations while batching in qblox. I'm not sure about the status for QM. Either way, I believe this issue to be more generic than the current state of the code and therefore it is not so useful anymore. Instead issues related to memory limitation should be about specific drivers, or even specific issues related to memory limiations within the qblox driver. 

For this reason I'm closing the issue. If you disagree, feel free to open again.

closed 2026-09-09 by RoyStegeman

alecandido, 2026-09-09:
No, you're right. This issue was aging anyhow, and the approach by now is completely superseded.

At the time, the goal was to minimize the driver-specific code, to abstract everything which was not strictly related to the drivers. Each driver was huge, and mostly for no reason (there were barely sweepers). This is not the case any longer, and we have more and more clear how such an interface should look like.

Instead, we now know that the unrolling and memory limitations are issues very specific to the individual drivers. And we are already dealing with them beyond what was initially proposed by this issue.