qiboteam/qibolab #716: Pulse deduplication

AleCandido, 2023-12-18:
This is already spanned by all (or most of) the drivers, but it is implemented in an inconsistent way.

- QM always requires the waveforms to be uploaded, and they could be unique, but at the moment they are not (mostly because of #682)
- Qblox has its own mechanism for deduplication implemented in the driver
    https://github.com/qiboteam/qibolab/blob/c6f41c2a115929dd5ccc9f2471870fbf25536665/src/qibolab/instruments/qblox/sequencer.py#L9-L12
    https://github.com/qiboteam/qibolab/blob/c6f41c2a115929dd5ccc9f2471870fbf25536665/src/qibolab/instruments/qblox/sequencer.py#L70-L73
- Zurich would benefit as well from something like this #711 

(the only one about which I know nothing is QICK) 

This became especially relevant after #618 

### Proposal

This feature could be implemented consistently across all drivers, and it will be pretty simple after the rework in #683 

The idea is to apply a strategy similar to Qblox one, consisting in a comparison:
- at the level of the waveform itself
- at the level of the parameters used to generate the waveform (if all the parameters affecting the waveform are the same, then the waveform will be the same)

If there is no other sensible equality for `Pulse` (something distinguishing `Pulse` with the same waveform) this will be their equality. Otherwise we could make a wrapper, just to define the equality, or store the waveform parameters in the `.waveform` attribute, and compare those.

Once we have objects that will compare equal if and only if the waveform is the same, we can just use the built-in mechanics of [`set()`](https://docs.python.org/3/library/stdtypes.html#set), and let Python deduplicate them.

This can be used simply in a series of contexts:
- deduplicating a single pulse sequence
- deduplicating an unrolled sequence
- deduplicating a swept sequence

It's just a matter of comparing the suitable pulses once they are generated.

AleCandido, 2023-12-18:
At least for Qblox, it might be slightly more complex than that.

Currently, the sequencers assigned to a port are playing a connected piece of a sequence. That's an extra constraint, and the motivation for the current pulse-assignment algorithm (try if it fits the memory of the current sequencer, otherwise try with the next one).

Deduplicating the whole sequence is not generally possible, because if pulses with the same waveform will land on different sequencers, they will have to uploaded multiple times.
However, if interleaved sequencers are possible (`[seq1] + [seq2] + [seq1]`) then we could deduplicate globally. Otherwise we could just apply the Qblox algorithm in general (or make it optional/enable for the case of multiple sequencers).

Do QM and Zurich have similar issues? @stavros11 @Jacfomg

Jacfomg, 2023-12-18:
I don't think Zurich works like that.