qiboteam/qibolab #747: Frozen objects

stavros11, 2024-01-11:
In particular, I would always replace, and make `QubitPair` frozen.

_Originally posted by @AleCandido in https://github.com/qiboteam/qibolab/pull/737#discussion_r1448494654_

Also for other objects: `Pulse` could be immutable if `start` is removed. `QubitPair` can be immutable if we replace `Qubit` with the qubit names.

For `Qubit` I am not so sure, because:
* it contains channels that we are setting during platform creation,
* it contains characterization parameters that qibocal is updating.

AleCandido, 2024-01-11:
> For `Qubit` I am not so sure, because:
> 
>     * it contains channels that we are setting during platform creation,
>     * it contains characterization parameters that qibocal is updating.

Indeed, we even discussed this briefly with @hay-k.

The current `Qubit` object is taking a lot of responsibilities, so it is a big deal everywhere it's used.

In my opinion, it would fit better for the second purpose (containing the characterization parameters, and potentially other relevant info, if ever), but avoiding taking action through the `Qubit` itself (which instead I'd delegate to just `Channel`s and `Port`s, where especially the second already contains device specific information).

stavros11, 2025-04-16:
`Pulse` and `Qubit` are already frozen in qibolab 0.2 as they inherit pydantic `Model`. `QubitPair` no longer exists. Therefore, I will close this issue as completed.

closed 2025-04-16 by stavros11