qiboteam/qibolab #757: Qblox SplitPulse

AleCandido, 2024-01-15:
Despite the lack of need in general for a `SplitPulse`, we might need it specifically for the Qblox driver (as originally pointed out in a comment to the implementation).

https://github.com/qiboteam/qibolab/blob/47d1d550a32f455fa2cc149820fa0ce06ae96795/src/qibolab/pulses.py#L1146-L1149

To the best of my understanding, this is motivated by the usage of a quite low-level interface in Qblox with respect to the other drivers, and the explicit presence of multiple sequencers (over which you might want to break even a single pulse).
This is not exposed by any other API, but it might also be handled internally.

Currently, it's not urgent, but better to keep track of the existing implementation and the potential need for it (or something similar).

PiergiorgioButtarini, 2024-01-17:
Thanks @AleCandido  for keeping track of this. Btw I can see the utility of splitting a single pulse in different sequencers only when this pulse is longer than 16834 ns (memory limit of each sequencer).

alecandido, 2024-09-06:
This will be solved by the Qblox rewriting, one way or the other.

closed 2025-02-19 by alecandido