qiboteam/qibolab #779: Adapt Qblox to work without `Pulse._if`

PiergiorgioButtarini, 2024-01-22:
> Since I should adapt a small part of the code in Qblox to work with this modification should I open a PR now and reference this PR or should I wait until all the pulse refactor is completed?

If you want to do it now, yes.

If you'd prefer to do it later, you will to use `0.2` (unless you want to wait a lot, such that `0.2` will be in `main`, but it will take some time and testing).

_Originally posted by @alecandido in https://github.com/qiboteam/qibolab/issues/768#issuecomment-1904305447_

alecandido, 2024-01-30:
Is this solved by #729? @PiergiorgioButtarini

PiergiorgioButtarini, 2024-01-30:
No it isn't, I still need to work on that.

alecandido, 2024-04-23:
The property is not available any longer. So, this will be automatically done by 0.2.0. No need for tracking it separately.

closed 2024-04-23 by alecandido