qiboteam/qibolab #785: Documentation update after release

stavros11, 2024-01-29:
Qibolab public stable documentation https://qibo.science/qibolab/stable/ is still for 0.1.4, even though we released 0.1.5 three days ago (https://pypi.org/project/qibolab/#history). @scarrazza @alecandido do you have any idea what went wrong?

scarrazza, 2024-01-29:
We have this issue in the past. We should relaunch the doc CI for the main branch and see if that works.

alecandido, 2024-01-29:
The answer is that we don't know. What happened before was that the workflow ran correctly, and the artifact to be published on Pages was good (you could check the artifact uploaded), but then it took a further run to update the actual website...

It should be somehow related to the way GitHub publishes things, but I did not manage to understand...

alecandido, 2024-01-29:
And indeed, that's the case once more...

[image]
https://github.com/qiboteam/qibolab/actions/runs/7670048477

andrea-pasquale, 2024-01-29:
After re-launching the workflow it seems to be working.
Looking here https://github.com/qiboteam/qibolab/actions/runs/7670048477 there are some warnings related to `node.js`. Maybe this is unrelated but could this be the cause?

alecandido, 2024-01-29:
> After re-launching the workflow it seems to be working.

It happened the exact same last time... still don't know why...

> Looking here [qiboteam/qibolab/actions/runs/7670048477](https://github.com/qiboteam/qibolab/actions/runs/7670048477) there are some warnings related to `node.js`. Maybe this is unrelated but could this be the cause?

I'm pretty much convinced it is unrelated. True that `upload-artifact` is among the outdated actions, but I doubt the problem to be just the Node version.

However, I will update the workflows soon (or @Edoardo-Pedicillo could do it as well, there is only to update the numbers, and check that the arguments we are using are all still there).

andrea-pasquale, 2024-01-31:
Quick update: now the documentation is rendered correctly (I relaunched the job a few days ago).
The only problem seems to be the button in the landing page.
[image]

scarrazza, 2024-01-31:
My fault, this is updated manually... fixed now.

closed 2024-01-31 by scarrazza